sparkfun/SparkFun_BNO080_Arduino_Library #2: Cannot upload arduino sketch on IMU(BNO080) through CH340G

s05vik, 2018-03-06:
With all the connections made and drivers installed, I am failing to upload arduino sketch on BNO080. 
In "Arduino.exe > Tools > 
                                      Board:  the default selected is 'Arduino/Genuino Uno'.
                                      Port: The detected one (COM9 in my case)
And then I try uploading. But fails to upload.

Note: CH340G tested and is working fine. (Tx/Rx loop back test)..Possibly it cannot recognize the board selected. Any help is appreciated. Thanks!!

nseidle, 2018-03-06:
Hi s05vik,

Perhaps there is some confusion. You cannot upload sketches to the BNO080 board. This library and the examples are designed to be uploaded to a RedBoard or Arduino Uno.

The BNO080 has a serial interface but is uses a different format that this library is not compatible with.

Cheers,
-Nathan

closed 2018-03-06 by nseidle

s05vik, 2018-03-07:
@nseidle  Perhaps the confusion lies in this link which suggests otherwise maybe :

_https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library_

nseidle, 2018-03-07:
Hi,

I am very sorry and do not want to mislead any customer. If you have a unit and want to return a product we are happy to give you a refund. Just let me know.

To answer your concern, I still don't see the text on the product page or the library page that would lead a customer to believe that this product is programmable with an Arduino sketch. 

The IMU080 is a sensor that is communicated with over various protocols including I2C, SPI, and serial. It has a serial interface. But that does not mean it can be programmed with Arduino.

Again, I'm really sorry about any confusion. Please help me write a better product description and I will change it.

Cheers,
-Nathan