sparkfun/SparkFun_BNO080_Arduino_Library #5: [Request] BNO080 SPI Library

ramiss, 2018-07-22:
Hi,  
Thanks for a great product. Unfortunately I need high precision 5-10 ms in a balancing robot that has 4 motors in close proximity.  As you may know the i2c interface/protocol is VERY susceptible to noise and then stopping and my balancing robot then... well... doesn't.

PLEASE can you write us an SPI library for the BNO080??

Even if you have a beta version (or another that is similar that I can adapt)  I'd be very interested in trying it.

Thanks!

nseidle, 2018-07-27:
SPI is a bit tricky to setup on the BNO080. Here's my working init code and notes. I'll try to fold this into the library soon.

[InitializeExample.zip](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/files/2236350/InitializeExample.zip)

ramiss, 2018-07-27:
Much appreciated!  I'll play around with this.

nseidle, 2018-07-27:
Lib manager should pick up v1.1.0 in a few hours. Let me know what you think!