sparkfun/SparkFun_BNO080_Arduino_Library #7: Getting linear acceleration data

ayu135, 2018-07-26:
Hi,
I was working with the bno080 sensor and wanted to get the linear acceleration data ie. the gravity subtracted acceleration but I wasn't able to use to getLinAccelX() function to get that data after calling the enableLinearAccelerometer() function. The arduino IDE doesn't even recognize the getLinAccel function and throws an error. 

After toiling around a bit i was able to get the data by modifying the report id for the normal accelerometer data to 0x04 in the macros in the header file of the library. I was hoping that instead of this work around we could use the normal linear acceleration functions to access that data.

nseidle, 2018-07-27:
I just pushed example 12 which shows how to use linear accel. Works for me. Let me know if you have any problems.

closed 2018-07-27 by nseidle

SHANTANUB009, 2018-07-28:
Hey, Works great for me. Also compared the hack that i tried, with your new push, both give identical results.   And Thanks for the SPI interface as well.

nseidle, 2018-07-28:
Very good! Happy to help.