sparkfun/SparkFun_BNO080_Arduino_Library #8: X and Y getting out of sync when Yaw and then settling back

ramiss, 2018-07-28:
Hi,

First of all I want to say a HUGE thank you for the SPI library mod. It works great!

Second, whether I2C or SPI I have this issue where the X and Y axis are thrown off by a few degrees when I rotate/yaw the IMU and then settle back to zero some seconds later.  Trust me when I say I am not tilting it on the x/y axis when doing this,  it is on a ballbot and when rotated the thing wants to fall over. Its off by over 6 degrees when rotated quickly 90 degrees at yaw.

Couple things:

1) I have been absolutely unsuccessful in calibrating the magnometer. No matter how many times I rotate around the axis and back it will constantly say unreliable.  However, I am hard pressed to believe that a badly calibrated magnometer will work perfectly until rotated slightly and then it "happens" to find center again.

2) I am converting from Quat to Euler with the following code.  Is this perhaps my issue?  I cannot figure out why the BNO080 doesn't have the option to convert to Euler.  My application does not have a worry of gimbal lock.

` ret_val.yaw = radiansToDegrees(atan2(2*(q.k*q.real+q.i*q.j),  q.i*q.i - q.j*q.j - q.k*q.k + q.real*q.real));
  ret_val.pitch = radiansToDegrees(-asin(2*(q.j*q.real-q.i*q.k)));
  ret_val.roll = radiansToDegrees(atan2(2*(q.j*q.k+q.i*q.real),  q.i*q.i + q.j*q.j - q.k*q.k - q.real*q.real));`

Any help is greatly appreciated, I'm hoping this is a code issue and not a bad board or bad hardware design.

Thanks!

Richard

ramiss, 2018-07-31:
Update,

I have been able to calibrate all sensors to a level of High and the issue still happens. I have also confirmed that the manipulation of the X and Y axis is only noticeable for the Quats generated by the BNO080. The raw values do not "seem" to have the issue, which would indicate a Quat firmware issue rendering this IMU pretty much useless.

Any thoughts are appreciated.

nseidle, 2018-07-31:
Hi Richard,

Your welcome! I'm glad you got the SPI interface working. It's a bit of a pain (many GPIOs).

We've written the library to pull data as best as possible. I can't vouch for the firmware that runs on the BNO080 itself. I don't mean to point fingers but I have to direct you to contact Hillcrest as they are the one ultimately in control of what data the device produces.

Sorry :/ I'm not helpful. I'm currently trying to get a contact from someone at SparkFun who met someone at Hillcrest so I can direct you to them. TBD.

ramiss, 2018-07-31:
Thank you. I've also reached out to Hillcrest and got hold of a contact. I'm in the process of providing more info to them. I'll report back here when I know more.

nseidle, 2018-08-07:
Sounds like Hillcrest is helping you. I wanted to record the main note from them:

* In general, calibration is automatically done by the BNO080 every few minutes. You do not need to explicitly enable calibration, nor should you *disable* calibration during normal operation.

I added a feature to the lib that checks the response of the ME Calibration command. See v1.1.1 that I just pushed as a release. I re-wrote the calibration example to include this check and *remove* the disable calibration function call. Let me know if you still have issues.

closed 2018-08-07 by nseidle

ramiss, 2018-08-07:
Excellent.  Thank you Nathan.  I will see if this helps.  My biggest issue is that even if I ran the previous calibration routing and saved it once they were all at "high" the magnetometer would be at 0 "unreliable" again once reset.  I'll test if this helps and get back to you.