sparkfun/SparkFun_BNO080_Arduino_Library #11: Receive Packet keeps returning in channel 0

andreydung, 2018-07-30:
I'm porting code to nordic nRF52832, and have been stucking at this problem for a while. Any help is much appreciated.

After sending a packet (either softReset or asking for chip ID), receive packet always returning the SHTP Header in channel 0. Here is an example of receivePacket output after sending 1 byte of 1 on CHANNEL_EXECUTABLE.

```
BNO080:INFO:SHTP HEADER: 14 81 0 fd
BNO080:INFO:data Length 276
bytesRemaining: 272 f9 0 0 0 0 0 0 0 e8 1f 0 0 0 0 0 0 0 0 0 0 0 0 0 0 d8 1f 0 0
bytesRemaining: 244 f9 0 0 0 0 0 0 0 e8 1f 0 0 0 0 0 0 0 0 0 0 0 0 0 0 d8 1f 0 0
bytesRemaining: 216 f9 0 0 0 0 0 0 0 e8 1f 0 0 0 0 0 0 0 0 0 0 0 0 0 0 d8 1f 0 0
bytesRemaining: 188 f9 0 0 0 0 0 0 0 e8 1f 0 0 0 0 0 0 0 0 0 0 0 0 0 0 d8 1f 0 0
bytesRemaining: 160 f9 0 0 0 0 0 0 0 e8 1f 0 0 0 0 0 0
```

I'm wondering if you have any clue regarding this?

Freefly18, 2018-07-31:
Hi andreydung. You can take a look at how Hillcrest does it here: https://github.com/hcrest/bno080-nucleo-demo

You can also contact Hillcrest support in order to get help from one of their engineer, as this seems more of a problem between your code and the BNO080, rather than what this library intends.

nseidle, 2018-07-31:
Thanks @Freefly18, I agree.

@andreydung - Sorry, this is a lib written for Arduino. I'm afraid I can't support an nRF port at this time. If you do figure out the issue, please let us (and everyon else) know.

closed 2018-07-31 by nseidle