sparkfun/SparkFun_BNO080_Arduino_Library #14: Timestamps

abdallahJabr, 2018-11-17:
Would it be possible to add timestamps to the reports returned for the rotation vectors, accelerations, etc? I read the entire datasheet and then the example code, but I wasn't able to modify the original functionality to add the timestamps reported by the sensor.

It would be super helpful to have them!

besabestin, 2018-11-21:
I second this comment. I tried to use `micros()` from arduino and I encountered issues, specially when collecting data from a lot of sensors.

nseidle, 2018-11-28:
Alright, [section 1.4.5.3](https://cdn.sparkfun.com/assets/1/3/4/5/9/BNO080_Datasheet_v1.3.pdf) talks about time stamping. Looking through the lib, I believe the timestamp is exposed in the [shtpData array here](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/src/SparkFun_BNO080_Arduino_Library.cpp#L202).

So I added:

    timeStamp = ((uint32_t)shtpData[4] << (8 * 3)) | (shtpData[3] << (8 * 2)) | (shtpData[2] << (8 * 1)) | (shtpData[1] << (8 * 0));

And exposed it via `getTimeStamp()`. Then I added an example 13 to show how to print it. Looks like it's working so I've pushed v1.1.2. Please kick the tires and let me know if it's what you're looking for.

[image]

closed 2018-11-28 by nseidle

besabestin, 2018-11-28:
I will check this out this week. But quick look on that snippet - what unit are you using? and considering `myIMU.enableRotationVector(50);` a 50ms this doesn't feel like in microseconds - unless ofcourse `myIMU.dataAvailable()` isn't affected by the sampling rate.

nseidle, 2018-11-28:
According to section 1.4.5.3 units are microseconds. And the time stamp is the time since the interrupt occurred (every 50ms) and when we move the bytes from the frame buffer into the global vars, then printed via the .getTimestamp() function. So ~30 micros seems reasonable.