sparkfun/SparkFun_BNO080_Arduino_Library #15: [Request] BNO080 UART library

m2oswald, 2018-11-30:
Hi there.  I was wondering if you have any sketches that communicate via UART?  The I2C sketches work great, but I need UART because my device will be located a few meters away from the Arduino (I'm using the Serial1 pins on a Mega).  I have been trying for a few days now but have gotten nowhere.  Any help/advice you could give would be GREATLY appreciated.  Thanks.

STM-Cortex, 2019-10-09:
I'd like to take it on

nseidle, 2019-11-15:
Great! Thank you! Please create a branch and let me know if you have any questions.

PaulZC, 2020-04-29:
Hi @m2oswald ,
Looking at the [BNO080 datasheet](http://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf), there are two ways the UART can be used:
- at 115200 baud when in Robot Vacuum Cleaner (RVC) mode
- at 3Mb/s for the full SHTP protocol

If RVC mode is all you need, you should be able to parse the data quite easily by following the protocol definition in the datasheet.
If you need the full SHTP protocol then, sadly, 3Mb/s is out of reach for most standard platforms.
I am going to close this issue as I don't think there is any way we can help you, but please reopen if we have missed something.
Best wishes,
Paul

closed 2020-04-29 by PaulZC