sparkfun/SparkFun_BNO080_Arduino_Library #16: Issue connecting ESP8266 Thing to BNO080

eskhisov, 2018-12-05:
I am trying to connect ESP8266 Thing, or any ESP8266 board for that matter, to BNO080 via I2C interface and it is not working. 
I have tried different things, including setting different parameters on Wire library, but nothing works.  Even when ESP8266 recognizes that there is a sensor attached, which seems to be about 70% of the time, the loop() function does not generate any data.  If I turn on debugging, I get a steady stream of 'I2C Timeout' messages.

eskhisov, 2018-12-14:
Ok, so I've got it to work, sort off.  See attached code, which is modified Example1-RotationVector.ino.
Caveats:
1.  ESP8266 Thing does work when I load the code for the first time.  Then, when I switch off and on, it mostly does not work, but does work sometimes.
2. I also tried it with generic ESP8266, which is this one https://www.amazon.com/gp/product/B010N1SPRK
With this one, it works almost always.  It does have reset button instead of on/off switch, but it failed maybe once out of 100 resets.

[sketch_nov23a.ino.txt](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/files/2678669/sketch_nov23a.ino.txt)

nseidle, 2019-01-23:
ESPs are notorious for their odd I2C issues. I can't work on this at the moment but we've found various solutions 

* Add a bit of time at beginning of the sketch before the myIMU.begin() to allow the BNO080 to boot
* Increaset the Wire.setTimeouts() before the myIMY.begin() to allow the BNO080 to clock stretch when needed

eskhisov, 2019-02-01:
I will try those suggestions.

DaStewie, 2019-03-28:
Having the following in the setup got it working for me, don't ask me why.
Connection:
SDA = 4 => D2.
SCL  = 5 => D1

Setup:
>	delay(100); // Allow BNO080 to startup
>
>	// Start i2c and BNO080
>	Wire.begin(5, 4); // idk why but removing this wouldnt work
>	IMU.begin(BNO080_DEFAULT_ADDRESS, Wire);
>	Wire.begin(4, 5); //
>	Wire.setClockStretchLimit(4000);

eskhisov, 2019-05-13:
@DaStewie Wow, it does work.  My setup is 2,14 but it works for it too:

`delay(100); //would not work without this delay
  Serial.println("BNO080 Read Example");

  Wire.begin(5,4);
  myIMU.begin(BNO080_DEFAULT_ADDRESS, Wire);
  Wire.begin(2, 14);              // (SCLpin,SDApin)
  Wire.setClockStretchLimit(4000);`

DaStewie, 2019-05-16:
Managed to find out what Wire.begin(5, 4) did. Using flush resets the index and allows for a proper startup.
`
  delay(100); //  Wait for BNO to boot

  // Start i2c and BNO080
  Wire.flush();   // Reset I2C
  IMU.begin(BNO080_DEFAULT_ADDRESS, Wire);
  Wire.begin(4, 5); 
  Wire.setClockStretchLimit(4000);
`

eskhisov, 2019-05-16:
This works too.  And much more rational.

nseidle, 2019-08-29:
Thanks you two! I've added this code (commented out) and a link to this issue in Example 1. Hopefully it will help future ESP users.

closed 2019-08-29 by nseidle