sparkfun/SparkFun_BNO080_Arduino_Library #17: Example1-RotationVector.ino Problem

da40flyer, 2018-12-08:
Hello,
I run the "Example1-RotationVector" SPI version of the sketch and the quaternion values spit out correctly on the serial monitor as expected.

If I add a simple delay(1000); in the loop after the Serial.println();  the quaternion values that are spit out to the serial monitor are now all zeros.

I've gone back and forth testing with and without the simple delay, and it is repeatable.

other tests:
If I use a delay(100); it appears to work fine
if I use a delay(500); nothing is sent to serial monitor
if I use a delay(1000); it sends zeros to serial monitor every second
If I remove the delay; all works fine again

Ideas on why?

ramiss, 2019-01-03:
The BNO080 communicates its readings on its own schedule and, if the Arduino isn't listening (delayed), then the readings are missed.  It is also bad practice to use a delay, instead your loop should be checking that 1000 seconds (or whatever the perceived delay you want) has passed and then do the println.

da40flyer, 2019-01-03:
I see.
Yes, I understand that delay is bad practice.  I simply used it to illustrate and test the problem I was seeing.

This problem rose its head because something as simple as adding or removing a serialPrint  had changed whether the data was captured or not.  Even a simple "if" statement would cause it to not read depending on the number of cycles it went through.

Anyway, thanks for replying.

nseidle, 2019-01-23:
I can replicate what you describe but I don't really see it as an issue.  The BNO080 is trying to output data but because of your delays, the data is piling up and the BNO080 is giving up. 

You should either increase the time for each vector, for example

    myIMU.enableRotationVector(1000); //Send data update every second

Or harvest the data more often.

closed 2019-01-23 by nseidle