sparkfun/SparkFun_BNO080_Arduino_Library #18: Rotation Vector + Accelerometer Data

vitorsotero, 2018-12-17:
Hi,

I want to get the Quaternion Data and the Accelerometer Data at the same time. 

I read the library and found this: 

"(shtpData[5] == SENSOR_REPORTID_LINEAR_ACCELERATION)" 

and 

"(shtpData[5] == SENSOR_REPORTID_ROTATION_VECTOR || shtpData[5] == SENSOR_REPORTID_GAME_ROTATION_VECTOR)"

So I am able to use just one or another? If so, how can I solve this?

Thanks,

Vitor Sotero

nseidle, 2018-12-18:
Have a look at the datasheet. You should be able to enable both reports and then scan for the incoming IDs. Do this by combining Ex1 and Ex2:

    myIMU.enableAccelerometer(50); //Send data update every 50ms
    myIMU.enableRotationVector(50); //Send data update every 50ms

then, once data is available:

    float x = myIMU.getAccelX();
    float y = myIMU.getAccelY();
    float z = myIMU.getAccelZ();
    float quatI = myIMU.getQuatI();
    float quatJ = myIMU.getQuatJ();
    float quatK = myIMU.getQuatK();
    float quatReal = myIMU.getQuatReal();
    float quatRadianAccuracy = myIMU.getQuatRadianAccuracy();

vitorsotero, 2018-12-20:
Sorry,

but when I try to enable both, the result is this: 
[image]

And if I just enable or Linear Accel or Rotation Vector, the result is like this: 
[image]

PS: the first figure I plotted only Linear Accel, but enabled Rotation Vector too.

On the "void Setup" I did:
myIMU.enableRotationVector(50); //Send data update every 5ms
myIMU.enableLinearAccelerometer(50); //Send data update every 5ms

And on the "void loop":
  float quatI = myIMU.getQuatI();
    float quatJ = myIMU.getQuatJ();
    float quatK = myIMU.getQuatK();
    float quatReal = myIMU.getQuatReal();
    float x = myIMU.getLinAccelX();
    float y = myIMU.getLinAccelY();
    float z = myIMU.getLinAccelZ();

//    Serial.print(quatReal, 2);
//    Serial.print(F(","));
//    Serial.print(quatI, 2);
//    Serial.print(F(","));
//    Serial.print(quatJ, 2);
//    Serial.print(F(","));
//    Serial.print(quatK, 2);
//    Serial.println();   

    Serial.print(x, 2);
    Serial.print(F(","));
    Serial.print(y, 2);
    Serial.print(F(","));
    Serial.print(z, 2);
    Serial.println();

Thanks!

nseidle, 2019-02-18:
Closing due to inactivity. Please re-open if you're still experiencing this issue.

closed 2019-02-18 by nseidle