sparkfun/SparkFun_BNO080_Arduino_Library #20: Connecting to two BNO080 on the same I2C bus

nsty, 2019-02-17:
I am trying to connect to two BNO080s on the same I2C bus. I am using ATMega328p hardware I2C with the Arduino Wire.h library. I disabled internal pull-ups that are set in the Wire.h library making the 5V chip compatible with the 3v3 I2C bus. As recommended in the Sparkfun Hookup Guide I disabled the data line pull-ups on one board.

When both devices are physically connected to the bus at the same time, I am able to connect to both separately using addresses 0x4A and 0x4B. But as soon as I create two instances of class BNO080, only the first initialized one is working. While the second instance still recognizes its chip on the bus at begin() it only gives zeros whenever being asked for updated values.

It seems like the instances are interfering. I can't explain this behavior by now after studying the class. There seems to be only blocking calls to the Wire.h library whenever an address is submitted and we're waiting for an answer to read. As far as I can say the Wire.h library's asynchronous calls are not used at all. I'm not using Arduino IDE but I tried to set up a minimal not-working example for anyone that wants to check on this: [Example14-MultipleChips.ino.zip](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/files/2872964/Example14-MultipleChips.ino.zip)

I read in some forums that for other chips it's possible to use the ADR pin in a similar way as an SPI chip select pin. This would mean an additional wire per chip of course. I'm trying to avoid this and also looking into the datasheet it should not work without resetting the chip every time I change address.
"The lower address bit of the device address is provided by the SA0 pin. This pin is sampled at reset and should be tied to either a logic high or logic low." (Datasheet BNO080 v1.3 p.14).

nseidle, 2019-02-18:
I am able to connect two I2C sensors over I2C; one with jumper open, other with jumper closed:

[image]

I've pushed an example 14 demonstrating this.

It's not perfect. I often see the sensors freeze up after a sketch has been uploaded. I attribute this to the sensor buffer piling up because the Uno is not harvesting the reports. Once the buffer is over-run the BNO080 fails to respond to .begin(). To treat this, I either power cycle my string of BNO080s by unplugging/plugging my Qwiic cable-bus of sensors, or I can remove the hard while(1);s from the example and the sensors will return to working.

Let me know what you experience. We should be able to get two sensors working together without problems.

nsty, 2019-02-21:
Thank you for your reply. It got me on the right tracks I think. The main problem is probably that my bus is working right at the corner of working at all due too a bus length of around 1m. So it happens on a regular basis that the sensors are freezing when I'm talking to both of them. I tried to lower I2C speed to make the bus more stable. But this probably results in the sensor buffer pilling up after a while as you said.
I will put the possibility on my board to power-cycle the IMUs from the microcontroller when they freeze. Anyway I will try to improve my setup to shorten the bus length.

I read about clock-stretching being a problem on BNO055 because it causes a really tight timing when pull-up is too weak. Especially they talk about putting a stronger pull-up on SDA than on SCL. [https://forums.adafruit.com/viewtopic.php?f=19&t=114404](https://forums.adafruit.com/viewtopic.php?f=19&t=114404)
Do you have pull-ups enabled on both boards?

Do you think it might be an enhancement to use the asynchronous calls in Wire.h library to not block the MCU when the bus is read and go down with the bus speed? I'm thinking about implementing this. I will report again about my progress.

nseidle, 2019-03-05:
There are two 2.2k pullups on the BNO080 in the lower left corner. They can be removed if necessaryby clearing the jumper. You can then attach any resistors you need.

[image]

You can also decrease the wire speed down to [12.5kHz](http://www.gammon.com.au/i2c) if that helps.

nseidle, 2019-06-25:
Closing due to inactivity. Please re-open if you've got new news.

closed 2019-06-25 by nseidle