sparkfun/SparkFun_BNO080_Arduino_Library #22: How to read raw data from the IMU

besabestin, 2019-03-14:
I changed the following lines as follows in the header file to be able to receive the raw data

```cpp
#define SENSOR_REPORTID_ACCELEROMETER 0x14
#define SENSOR_REPORTID_GYROSCOPE 0x015
#define SENSOR_REPORTID_MAGNETIC_FIELD 0x16
```

However, `dataLength` is always 0. If I change the the report id for the accelerometer to 0x01 it works fine. I wanted to receive raw data from the device. What can I do?

nseidle, 2019-06-25:
Hmm, I attempted to implement this in [this](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/tree/Raw_MEMS_Values) branch. I had X working for a little while and something broke. I suspect the MEMS sensors are reporting at a much higher rate and when we don't harvest the reports, the sensor overflows its buffers and locks up.

Or it might be that I'm not parsing the timestamp located at bytes 12-15 (different from regular accel/sensor reports).

[image]

nseidle, 2019-06-25:
I found the issue, and I think it would be yours as well: I found you need to enable the accelerometer as well as rawAccelerometer, otherwise it never reports. This makes sense,

    myIMU.enableAccelerometer(50); //We must enable the accel in order to get MEMS readings even if we don't read the reports.
    myIMU.enableRawAccelerometer(50); //Send data update every 50ms

Branch now works. I'll pull it in and push a new version in a minute.

closed 2019-06-25 by nseidle