sparkfun/SparkFun_BNO080_Arduino_Library #23: Can't get example to work on SAMD21

simon-rob, 2019-04-27:
I have tried the Example - Rotation Vector.ino on a Genuino Uno and it worked flawlessly.

However if I try the same code on an Adafruit feather M0 proto or Adafruit Circuit Playground Express it just hangs. 

Enabling  myIMU.enableDebugging(Serial) shows  "I2C timeout".

I have tried adding a startup delay to give the BNO080 time to start up but still the same.

I am confused why it doesn't work on both SAMD21 boards!

simon-rob, 2019-04-29:
Answer my own question:

Commenting out Line 55 /56  in SparkFun_BNO080_Arduino_Library.h and it works.

//SAMD21 uses RingBuffer.h
//#define I2C_BUFFER_LENGTH SERIAL_BUFFER_SIZE

nseidle, 2019-06-25:
Thanks for coming back to report! Hmm, I wonder how SERIAL_BUFFER_SIZE is defined in the SAMD core that you're using and what's changed. I'll try to test on our core shortly.

nseidle, 2019-08-29:
I suspect Adafruit is change how the ring buffer is defined. I've removed the define from the library. The catch all define gate will work for SAMD21.

closed 2019-08-29 by nseidle