sparkfun/SparkFun_BNO080_Arduino_Library #24: Compilation warnings

MichaelPriven, 2019-05-06:
in file - SparkFun_BNO080_Arduino_Library.cpp, 
Line 233:
timeStamp = ((uint32_t)shtpData[4] << (8 * 3)) | (shtpData[3] << (8 * 2)) | (shtpData[2] << (8 * 1)) | (shtpData[1] << (8 * 0));
should be - 
timeStamp = ((uint32_t)shtpData[4] << (8 * 3)) | (**(uint32_t)**shtpData[3] << (8 * 2)) | (**(uint32_t)**shtpData[2] << (8 * 1)) | (**(uint32_t)**shtpData[1] << (8 * 0));

Line 1005:
int16_t dataLength = ((uint16_t)packetMSB << 8 | packetLSB);
should be -
**uint16_t** dataLength = ((uint16_t)packetMSB << 8 | packetLSB);

nseidle, 2019-06-25:
Thank you for the recommended correction! I am not seeing any warnings on AVR(Uno) or Cortex-M4F (Artemis core). What development board or microcontroller are you using so I can replicate the issue?

MichaelPriven, 2019-06-26:
I'm using Arduino Mega2560 and Visual Studio with visual micro

nseidle, 2019-08-29:
Can't replicate. Perhaps Visual micro changes the compiler flags because I'm not seeing these.

Should be fixed in next version.

closed 2019-08-29 by nseidle