sparkfun/SparkFun_BNO080_Arduino_Library #25: NULL compilation warning

MichaelPriven, 2019-05-06:
Hi,

I suggest to change ...**uint8_t intPin = NULL** to uint8_t intPin = 255 to avoid compilation warning / error (in new version of Arduino IDE)

Thanks
Michael

nseidle, 2019-06-25:
Thanks! What microcontroller are you using? What version of the IDE are you using? I can't replicate on Uno or Artemis in Arduino v1.8.9.

MichaelPriven, 2019-06-26:
I'm using Arduino Mega, with V1.8.9 or visual studio with visual micro

nseidle, 2019-08-29:
Oddly I can't replicate. Windows 10, Arduino v1.8.9. Have you modified your platforms.txt? It sounds like the warnings on Arduino have been tweaked up in your version.

Regardless, I've changed the problem line of code and should roll with then next lib version.

Thanks for reporting!

closed 2019-08-29 by nseidle