sparkfun/SparkFun_BNO080_Arduino_Library #27: Available Data

Belayasmert, 2019-06-17:
Hello,
thank you for the library and the code. It works fine for me. But i have an issue. After a couple of minutes the Sensor stops delivering data. Could it be that he went in to the sleep mode? I am not sure and it would be perfect for me when the sensor sends data without any break.

nseidle, 2019-06-17:
You're welcome! Thanks for using this library.

If you would please answer a few questions:

### Your workbench
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

Belayasmert, 2019-06-18:
Ok there was to little information Sorry.
I am using an Arduino Nano,  the BNO080 is on a Breakoutboard which i created by myself. I followed the I2C reference design from the BNO080 Datasheet. This breakoutboad has a Pinheader with the Pins 

**BNO_BOOTN, GND, BNO_CLKSEL0, BNO_NRST, HOST_INTN, +3V3, HOST_SCL, HOST_SDA**

My arduino Nano is powered by a usb cable, the Breakoutboard is just connected with the power supply pins and the I2C pins. 

I used the Example1-Rotationvector code with the SparkFun_BNO080_Arduino_Library.  Only thing i changed is 
`Wire.setClock(100000);`
Because 400000 didn't work for me, the Arduino couldn't find the BNO080 on the I2C line.

In the loop i just added the else code.

```
  //Look for reports from the IMU
  if (myIMU.dataAvailable() == true)
  {
   ....
   ....
   ....
    Serial.println();
    i=0;
  }
  else
 {
    Serial.println("keine neue daten");
  i++;

  if ( i >= 10 )
  {
   Serial.println("more than 10 tries");
   i = 0;
  }
 }
```
to see the behavior of the loop and the ouput.

After i Power up my Arduino the programm starts without any problems !
The output on the serial monitor looks like this.

```
09:19:20.197 -> BNO080 Read Example
09:19:20.431 -> Rotation vector enabled
09:19:20.478 -> Output in form i, j, k, real, accuracy
09:19:20.525 -> 0.00 0.00 0.00 0.00 0.00
09:19:20.525 -> keine neue daten
09:19:20.572 -> keine neue daten
09:19:20.572 -> keine neue daten
09:19:20.572 -> keine neue daten
09:19:20.619 -> keine neue daten
09:19:20.619 -> keine neue daten
09:19:20.619 -> keine neue daten
09:19:20.666 -> keine neue daten
09:19:20.666 -> 0.00 0.00 0.00 0.00 0.00
09:19:20.713 -> 0.28 0.24 -0.72 0.59 0.79
09:19:20.713 -> keine neue daten
09:19:20.759 -> 0.28 0.24 -0.72 0.59 0.79
09:19:20.759 -> keine neue daten
09:19:20.806 -> keine neue daten
09:19:20.806 -> 0.28 0.24 -0.72 0.59 0.79
09:19:20.853 -> keine neue daten
09:19:20.853 -> 0.28 0.24 -0.72 0.59 0.79
09:19:20.900 -> keine neue daten
09:19:20.900 -> 0.28 0.24 -0.72 0.59 0.79
09:19:20.947 -> keine neue daten
```
After about 15 mins the Sensor stoped to deliver data.

```
09:34:36.290 -> 0.33 0.21 -0.79 0.48 0.79
09:34:36.290 -> keine neue daten
09:34:36.290 -> 0.33 0.21 -0.79 0.48 0.79
09:34:36.337 -> keine neue daten
09:34:36.337 -> 0.33 0.21 -0.79 0.48 0.79
09:34:36.431 -> keine neue daten
09:34:36.525 -> keine neue daten
09:34:36.618 -> keine neue daten
09:34:36.665 -> keine neue daten
09:34:36.712 -> keine neue daten
09:34:36.712 -> 0.33 0.21 -0.79 0.48 0.79
09:34:36.712 -> keine neue daten
09:34:36.759 -> keine neue daten
09:34:36.759 -> keine neue daten
09:34:36.806 -> keine neue daten
09:34:36.806 -> keine neue daten
09:34:36.806 -> keine neue daten
09:34:36.853 -> keine neue daten
09:34:36.853 -> keine neue daten
09:34:36.900 -> keine neue daten
09:34:36.900 -> keine neue daten
09:34:36.900 -> more than 10 tries
09:34:36.946 -> keine neue daten
```
And thats it. I tried to reset the BNO080, but it doesn't work, i am not sure whats going on there.

Best Regards
Walter

nseidle, 2019-06-25:
I'll leave my BNO080 running overnight to try to replicate your issue but without that let me guess at a couple things:

1) Hardware issue. You've spun your own PCB (congrats!) but I can't help troubleshoot power, soldering, etc issues.
2) By decreasing the I2C clock to 100kHz you may not be harvesting the accel data fast enough. The BNO080 will freeze if its buffer overflows.

nseidle, 2019-06-26:
FWIW I ran our board over night. Not hiccups or freezes for 12+ hours.

Belayasmert, 2019-06-27:
Thanks for this information! I will try to increase the clock and make some further tests!

nseidle, 2019-08-29:
Closing issue due to age and inactivity. Please open a new issue if needed.

closed 2019-08-29 by nseidle