sparkfun/SparkFun_BNO080_Arduino_Library #29: How to avoid polling?

vitorsotero, 2019-07-10:
Hi,

there is this sentence on examples from the library: "Use the interrupt pin on the BNO080 breakout to avoid polling."

I tried to connect the INTpin to GND, but the results didn't change. Do I have to modify some line of the library?

Thanks!

nseidle, 2019-07-14:
You should not connect the INT pin to ground, you should connect the INT pin to your micro. When the INT pin on your micro goes low, you know there is new data available and you should request the new data from the BNO080.

closed 2019-07-14 by nseidle

DanieleSgroi, 2019-07-30:
found that using i2c with interrup pin polling, I need to add "pinMode(_int, INPUT_PULLUP);"

just before "softReset();" inside "BNO080::begin" function, otherwise the code get stuck.

However this is not a clean interrupt management, in the end it still is a pin polling.

I'm working to convert it to proper ISR...

vitorsotero, 2019-07-30:
> found that using i2c with interrup pin polling, I need to add "pinMode(_int, INPUT_PULLUP);"
>
> just before "softReset();" inside "BNO080::begin" function, otherwise the code get stuck.
>
> However this [...]

Interesting...If you do, please reply here! I'd like to test!

BNO055 was a little easier to request two or more functions (i.e. linear acceleration, gyroscope and quaternion).