sparkfun/SparkFun_BNO080_Arduino_Library #32: Support for Gyro-Integrated Rotation Vector

Nate711, 2019-09-17:
Just creating this issue to notify you that I'm working on adding support for the gyro-integrated rotation vector input report. This 'sensor' reports an absolute rotation quaternion and angular velocity at up to 1000Hz which I think could help many people using this sensor for robots with fast dynamics such as drones or legged robots.
https://github.com/nate711/SparkFun_BNO080_Arduino_Library. I'll make a PR once testing is done. Thanks for your work!

vitorsotero, 2019-09-19:
I gave it a try and got this compilation error:

`C:\Users\Sotero\Documents\Arduino\libraries\SparkFun_BNO080_Arduino_Library\src\SparkFun_BNO080_Arduino_Library.cpp:508:28: error: no 'float BNO080::getFastGyroX()' member function declared in class 'BNO080'

 float BNO080::getFastGyroX()

                            ^

C:\Users\Sotero\Documents\Arduino\libraries\SparkFun_BNO080_Arduino_Library\src\SparkFun_BNO080_Arduino_Library.cpp:515:28: error: no 'float BNO080::getFastGyroY()' member function declared in class 'BNO080'

 float BNO080::getFastGyroY()

                            ^

C:\Users\Sotero\Documents\Arduino\libraries\SparkFun_BNO080_Arduino_Library\src\SparkFun_BNO080_Arduino_Library.cpp:522:28: error: no 'float BNO080::getFastGyroZ()' member function declared in class 'BNO080'

 float BNO080::getFastGyroZ()

                            ^

C:\Users\Sotero\Documents\Arduino\libraries\SparkFun_BNO080_Arduino_Library\src\SparkFun_BNO080_Arduino_Library.cpp:830:76: error: no 'void BNO080::enableGyroIntegratedRotationVector(uint16_t)' member function declared in class 'BNO080'

 void BNO080::enableGyroIntegratedRotationVector(uint16_t timeBetweenReports)

                                                                            ^

exit status 1
Erro compilando para a placa Arduino Nano`

Nate711, 2019-09-27:
Thanks for pointing that out! Please try again with the new code.

vitorsotero, 2019-10-01:
Now I can compilate, but it doesn't show any results. 

When I call "myIMU.enableGyroIntegratedRotationVector(1);", it doesn't change the color to orange (arduino code). It differs from the other functions (ex: myIMU.getQuatI();)

nseidle, 2020-01-22:
Very cool! Please continue your work. We'd welcome a PR to add more features.

spiderolsen, 2020-04-05:
> I managed to make this work and made a pull request (my first ever \o/). I have to give credit to Nate711 because I started from one of his old commits. Don't know how to pull in old stuff so the chain would be intact. The code is not in Nates current branch:/.
> 
> Anyway, it works and its fast.

Hi badVibes,

That are pretty good news! I´m currently working on a DIY VR Project and I´m really interested in using the gyro-integrated Rotation vector, because its low latency claim. I´ve already changed the **#define SENSOR_REPORTID_ROTATION_VECTOR** to **0x2A**, but the output remains at 0 for all Quaternion values. Is there any chance for sharing your solution or any tips? Thank you in advance and stay healthy!

spiderolsen, 2020-04-05:
> I managed to make this work and made a pull request (my first ever \o/). I have to give credit to Nate711 because I started from one of his old commits. Don't know how to pull in old stuff so the chain would be intact. The code is not in Nates current branch:/.
> 
> Anyway, it works and its fast.

Hi badVibes,

I´ve just tested your rewritten library. Unfortunately with your changes it is not possible to turn town **myIMU.enableGyroIntegratedRotationVector(5);** below 5ms without freezing an arduino nano. With the normal sparkfun library it is possible to send a update every 1ms and the loop void runs about 250Hz. I think an arduino nano will crash as soon as the loop void is faster than around 250Hz. At **myIMU.enableGyroIntegratedRotationVector(5);** the loop void is runnging at 190-200Hz.

badVibes--, 2020-04-07:
> I´ve just tested your rewritten library. Unfortunately with your changes it is not possible to turn town myIMU.enableGyroIntegratedRotationVector(5); below 5ms without freezing an arduino nano. With [...]

Hey spiderolsen, that seems strange. One thing that comes to mind is that the nano might be too slow. I tested all this on teensy 3.2 & 3.6 which are quite a bit faster. If you request a 1ms update for any other rotation vector it will not send it any faster than 400hz. How are you interfacing  with the sensor, I2C or SPI?

PaulZC, 2020-04-29:
Hi @Nate711 & @badVibes-- ,
I've just merged PR #41 which adds support for the Gyro-Integrated Rotation Vector.
I'm going to close this issue but please feel free to reopen it if we have missed anything.
Best wishes,
Paul

closed 2020-04-29 by PaulZC