sparkfun/SparkFun_BNO080_Arduino_Library #34: timestamp and failing to start

yo90bosses, 2019-10-28:
the .getTimeStamp() funktion is giving me numbers between 8 and 20. i was expecting to get the time of the data since start in milliseconds. Is this an issue or do i not understand the number its giving me? 
another issue is im running the sensor over the SPI bus on an esp32 and if i try to start it to fast it failes to start. My code immediatly tries to start the BNO080 after it fails but it wont work unless i add a delay of 1.5s between tries. The reason i need this is because the sensor will only start after a second try.

chengdaqian, 2019-11-10:
I'm running into the same problem. Do you have any idea on how to interpret this? I also just get "random" number between 10 and 30.

nseidle, 2020-01-22:
Timestamp is described [here](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/src/SparkFun_BNO080_Arduino_Library.cpp#L214) and in the datasheet. It's the number of microseconds since the Cortex-M0 took the calculation and when you read it. This helps us align data, not to when it was read but when it was taken.

Not sure about the SPI + ESP32 issue. Library is working very well over SPI on Uno, Artemis, and Teensy for me.

closed 2020-01-22 by nseidle