sparkfun/SparkFun_BNO080_Arduino_Library #36: BNO080 IMU data freezing issues

ashish1brash, 2019-12-19:
I'm using BNO080 sensor to read IMU data and store it in log files onto an SD card. Everything works fine, except that the data freezes around 20Hz.It creates a problem for me because I'm trying to log IMU data at 20Hz and use other sensor capabilities at different frequencies.

Has anyone come across a similar problem or can anyone help me sort this issue?

AndyEngland521, 2019-12-30:
@ashish1brash what are you using to log data to an SD card? To me it seems like your bottleneck here might be the time it takes to log data to the SD card.

To test this I would make a dummy message that is the same size as your IMU data and attempt to send that at higher data rates to see if the bottleneck is your SD write or the IMU.

nseidle, 2020-01-22:
Please fill out the issue template so we can help you more. If you're using the BNO080 in SPI mode, that would be very telling.

### Subject of the issue
Describe your issue here.

### Your workbench
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

closed 2020-01-22 by ashish1brash