sparkfun/SparkFun_BNO080_Arduino_Library #37: Bno080 detected but fails at dataAvailable()

ionidestiny, 2020-01-21:
### Bno080 detected but fails at dataAvailable()
My Bno080 fails right from the start but my Bno080 was detected on setup.

### Your workbench
* ESP32 DevKit-Lipo
* GY-Bno080 custom
* Connect via I2C reference Design from Bno080 datasheet
* Powered by : LiPo battery / USB 

Pins
VCC   -----> ESP32 3.3V
GND  -----> ESP32 GND
SCL    -----> ESP32 PIN 22
SDA   -----> ESP32 PIN 21
AD0
CS
INT
RST
PS1   -----> GND
PS0   -----> GND

### Steps to reproduce
It fails on startup.

### Expected behavior
Bno detected and dataavailable() should return true to allow for collection of data

### Actual behavior
Bno detected but dataavailable() fails.

[image]

nseidle, 2020-01-22:
We only have the resources to support our hardware so I can only guess:

* I forget but I think the BNO080 uses clock stretching. I think the [ESP32 has problems](https://github.com/espressif/arduino-esp32/issues/81) with I2C stretching.
* Missing pull up resistors

closed 2020-01-22 by nseidle

ionidestiny, 2020-02-03:
hi @nseidle 
i've added a pull up resistor but im not entirely sure how to implement clockstretching for ESP32. 

I stumbled  upon this issue https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/16 but the esp32 wire.h does not include a function for wire.setClockStretchLimit(); 

is there a way i can stretch the clock limit for esp32.

thanks