sparkfun/SparkFun_BNO080_Arduino_Library #38: Is it possible to get somenthing about the covariance matrix?

JackFrost67, 2020-02-13:
### Subject of the issue
Hi, I'm working on a multi-imu system and I want to get from the BNO080 some informations about the covariance. I've read that every output package have a "status" byte, but what this byte stands for? Am I able to get some information about the covariance from byte?

### My workbench
* STM32-F767zi using STM32Duino 
* SPI communication to the master
* Powered from the master (STM32-F767zi)

JackFrost67, 2020-02-17:
Metadata is the answer

JackFrost67, 2020-02-20:
If a Kalman filter is applied on the board, is there a way to get the covariance matrix of the estimate status? Metadata gives a Q point 2 value which I don't understand what that means.

PaulZC, 2020-04-29:
Hi @JackFrost67 ,
I have looked all the way through the Hillcrest/CEVA [SH-2 Reference Manual](https://www.ceva-dsp.com/resource/sh-2-reference-manual/) and I don't see any way of accessing the covariance matrix.
Q point 2 provides a measure of the sensor bias (offset) or accuracy. Sadly that is not what you are looking for.
I am going to close this issue as I don't think there is any way we can help you, but please feel free to reopen if we have missed something.
Best wishes,
Paul

closed 2020-04-29 by PaulZC