sparkfun/SparkFun_BNO080_Arduino_Library #49: Wrong angle?

phhoef, 2020-05-18:
I mounted the breakout board on a wood plate.
Currently, I am only interested in the roll angle. I am using the EulerAngles example.
When rotating over the x axis, I get suitable angles.
But, when rotating over the y axis (to 90°), the roll angle becomes ~60°.
This seems wrong. Can someone explain why this is happening?

PaulZC, 2020-05-18:
Hi @phhoef ,
Can you please try calibrating the sensor. Do you still see the same issue?

[image]
[image]
[image]

Best wishes,
Paul

phhoef, 2020-05-18:
@PaulZC I am quite new to the BNO sensors ...
I have another sketch for calibrating the sensors - basically the same as the example.
As I understand, when pressing 's' the calibration data is saved to the BNO IC. I actually did this once in the mounted position. Not sure, if I made things worse as I just found the "angle problem".

Anyhow, I can try re-calibrating and will share my results ...

---
It's a little bit off topic.
The BNO calibrates itself to the real 0° level, right? I do have some experience with the MPU6050 and with this IMU the calibration was like getting the offsets when placed on a leveling device.
Is there already something similar built-in to the BNO to even out an offset?

PaulZC, 2020-05-18:
Hi @phhoef ,
You can find more details about the calibration procedure here:
https://www.ceva-dsp.com/resource/bno080-bno085-sesnor-calibration-procedure/
Best wishes,
Paul

phhoef, 2020-05-18:
Hallo @PaulZC Thanks for your help.
I assume that this is the same document as linked here: https://cdn.sparkfun.com/assets/9/e/1/d/9/Sensor-Calibration-Procedure-v1.1.pdf

I uploaded the example calibration sketch and moved the sensor until all indicators were "HIGH". Then I saved the calibration by pressing 's'.

After resetting and starting over, I noticed that the calibration status of each sensor was "MEDIUM". Is that correct? I would have expected, that the calibration is automatically loaded when powering on? Though, the overall calibration status is still "HIGH".

After the calibration, I uploaded the Euler example and got the same result.
When the board is flat on ground:
`-0.1,0.1,-170.6`

When gravity is pushing against x axis:
`72.1,-86.7,110.5`

When gravity is pushing in direction of y axis:
`-88.9,-4.4,-163.6`

PaulZC, 2020-05-18:
Hi @phhoef ,
The calibration procedure on the CEVA website is newer (Rev. 1.3) and covers the BNO085 too.
I'm trying to replicate what you are seeing. Are you sure this isn't just gimbal lock? When the board is pitched up at 90 degrees, the roll becomes meaningless? There's a good explanation here (see section 6):
http://www.chrobotics.com/library/understanding-euler-angles
Best wishes,
Paul

phhoef, 2020-05-18:
Hallo @PaulZC,

good point, thanks. I noticed, that the pitch angle increases slightly when I manually increase the roll angle.
For example: I am holding the plate at 45° roll angle and get following result:
`45.4,2.9,165.8`

Same, for the other axis. Increasing pitch angle and also roll angle will increase.
`-3.2,45.3,164.1`

When arriving ~90° in the axis, the other angle jumps from ~3° to 70°.
```
-0.4,0.1,-156.4
-0.3,-0.4,-156.2
-0.3,-1.2,-156.2
-0.2,-2.4,-156.1
-0.1,-3.9,-155.9
0.0,-5.4,-155.9
0.1,-6.8,-155.9
0.2,-8.3,-156.0
0.3,-10.2,-156.0
0.5,-11.9,-157.1
0.6,-13.6,-157.7
0.7,-15.1,-157.8
0.8,-16.1,-158.1
0.9,-17.2,-158.4
0.9,-18.1,-158.6
1.0,-19.1,-158.6
1.1,-20.1,-158.7
1.1,-21.0,-158.8
1.2,-21.9,-158.8
1.3,-22.8,-158.8
1.4,-23.7,-158.9
1.4,-24.5,-158.9
1.5,-25.3,-158.9
1.5,-26.1,-158.9
1.6,-27.0,-158.9
1.7,-28.2,-159.0
1.8,-29.6,-159.1
1.9,-30.8,-159.1
2.0,-31.9,-159.1
2.1,-33.0,-159.1
2.1,-34.0,-159.1
2.2,-35.1,-159.2
2.3,-36.1,-159.3
2.4,-37.0,-159.4
2.4,-37.9,-159.6
2.5,-39.0,-159.7
2.6,-40.3,-160.0
2.8,-41.8,-160.3
2.9,-43.3,-160.5
3.1,-44.7,-160.7
3.2,-45.9,-160.8
3.3,-46.8,-160.9
3.4,-47.9,-161.2
3.5,-49.0,-161.3
3.7,-50.1,-161.4
3.8,-51.3,-161.6
4.0,-52.6,-161.8
4.2,-54.1,-162.0
4.5,-55.7,-162.4
4.7,-57.2,-162.8
5.0,-58.7,-163.3
5.3,-60.0,-163.6
5.5,-61.2,-163.9
5.8,-62.2,-164.2
6.0,-63.2,-164.3
6.3,-64.3,-164.5
6.7,-65.5,-164.6
7.2,-66.9,-165.0
7.6,-68.2,-165.5
[... 344 lines cut ...]
81.0,-87.0,121.0
81.0,-87.0,121.0
81.0,-87.0,121.1
81.1,-87.0,121.0
81.0,-87.0,121.1
81.0,-87.0,121.0
81.0,-87.0,121.0
81.1,-87.0,121.0
81.0,-87.0,121.1
80.9,-87.0,121.2
80.9,-87.0,121.2
80.8,-87.0,121.3
80.9,-87.0,121.2
81.0,-87.0,121.2
81.0,-87.0,121.2
81.2,-87.0,121.0
81.3,-87.0,120.9
```

Is this gimbal lock? Not sure, how to recognize it ...
The quaternions are obviously not affected by gimbal lock. I am using the angle for height and distance calculations (I am measuring the hypotenuse in addition).
My valuation for the roll angle is -20 to 80°... 

Thanks for pointing out, that the your link to the calibration procedure is newer.
Do you know, if there is also a newer version of the data sheet?

PaulZC, 2020-05-18:
Hi @phhoef ,
I think you are possibly seeing two things:
- Gimbal lock at the high pitch angles
- Plus possibly the fact that your BNO080 axes are not precisely aligned with your rotation frame

If the BNO080 is rotated slightly, either physically or through calibration, then as you increase the pitch you will see some error (or 'bleed through') appearing in the roll.
My advice would be to read the three accelerometer values and calculate the roll manually.

Yes, there is a new version of the datasheet too:
http://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf

I don't think this is a problem with the library and I don't think I can help you any further. The IMU forum is the best place to ask for advice about sensors:
https://forum.sparkfun.com/viewforum.php?f=83
I am going to close this issue, but please reopen it if you are convinced it is a bug in the library.
Very best wishes,
Paul

closed 2020-05-18 by PaulZC

phhoef, 2020-05-18:
@PaulZC thank you very much for your input and impressions :-)