sparkfun/SparkFun_BNO080_Arduino_Library #50: SPI Data Throughput is very slow

ManeeshSingh1920, 2020-05-25:
### Subject of the issue
I am using SPI rotation Vector Example and In that, it is using 50ms time delay between reports... 50ms means approx 20Hz and I tested if I decrease this time delay below 20ms which is 50Hz then BNO not able to send data properly..  
But if I test it with I2C interface then I am able to set time delay 3ms which is 333hz which is very High compare to SPI interface..... 

but in practice SPi is set at 3Mhz and I2C is set as 400Khz so SPI data rate must be high but BNO is not able to give data rate more than 50Hz in case of SPI...

Is this problem with BNO080 or Sparkfun library.... becoz BNO datasheet says rotation vector conversion speed is 400Hz...
So If I can get 333Hz from I2C then why not from SPI.. 

### Your workbench
I am using the Arduino board also I tested with the STM32 Nucleo 401RE board.

### Steps to reproduce
I am doing a connection as per this library and just compiling and running it.. not doing any extra changes. 

### Expected behavior
SPI data rate must be high 

### Actual behavior
SPI data rate we are getting very low.. 

NOTE : - I am not doing any changes in example given or in Library provided... I am using it as it is.

PaulZC, 2020-05-25:
Hello @ManeeshSingh1920 ,
I can confirm that I have had the BNO080 running successfully using a 10ms time delay on SPI with an Arduino Uno. I suspect the problem is linked to your SPI interface or your Nucleo board.
Can you please check that you have all the SPI connections connected correctly and have configured the split pads correctly too?
Do you have an Arduino Uno that you can try?
Best wishes,
Paul

ManeeshSingh1920, 2020-05-25:
Hi PaulZC,

Thanks for your reply.

Actually I am not doing any manual configuration for Nucleo... I am using the nucleo library for STM32 as given in this link => https://github.com/stm32duino/wiki/wiki/Getting-Started
 
With this I am able to use the Arduino library on STM32 Nucleo board.... I don't need to do any extra setting or configuration....  It just works as an Arduino board only... 

so I am not doing configuration for SPI or I2C.. It is using Arduino's library for SPI and I2C interface... 
 
Since I am able to use this library code on 50ms and 20ms with SPI on my STM32 Nucleo board so it means SPI Interface is working properly.... And Ideally 10ms and 5ms time delay also should work .... because I am not doing any manual setting for SPI or I2C..... 
but unfortunately, it is not happening... I am not able to get data below 20ms delay.
	 
Can you confirm if you are able to get data at or above 300hz means 4-5 ms time delay also on Arduino Uno for Enable Rotation Vector Example?

And can we use Arduino Uno for BNO080 ? 
because this BNO080  library says Uno is 5V so not recommended to use it.

Thanks for your help and support,

Regards,
Maneesh

PaulZC, 2020-05-25:
Hello @ManeeshSingh1920 ,
Apologies. I am using a SparkFun RedBoard (ATmega328P) which can be configured for 3.3V (instead of 5V). I mistakenly referred to it as a "Uno".
Please check you have configured the BNO080 split pads correctly. Have you cut the links on the the I2C split pads on the top of the board (to remove the pull-up resistors)? Have you soldered the PS1 split pad on the bottom of the board?
Have you connected all nine wires correctly?
- 3.3V
- GND
- SCK
- SO
- SI
- CS
- WAK
- RST
- INT
Best wishes,
Paul

ManeeshSingh1920, 2020-05-25:
Hi Paul, 

Thanks for your reply.. 

I did the connection as suggested. Plz check the attached Pic and logs on 50ms

 
[image]

[image]

[image]

PaulZC, 2020-05-25:
Hi @ManeeshSingh1920 ,
You have the PS0 split pad shorted. Please clear the solder and try again. The PS0 jumper must be open.
Best wishes,
Paul

ManeeshSingh1920, 2020-05-25:
Hi Paul,

Thanks for your reply..

I did the changes as suggested by you but unfortunately on 10ms after giving few logs the sensor stops reporting the new data. Plz check the attached Pic and logs on 10ms and 50ms

[image]

[image]

[image]

[image]

[BNO_10ms.txt](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/files/4677644/BNO_10ms.txt)

PaulZC, 2020-05-25:
Hello @ManeeshSingh1920 ,
How long are you delaying for in the main loop?
For 10ms, you will need to reduce the delay to (e.g.) ```delay(5);``` otherwise you will see issue #42
Best wishes,
Paul

ManeeshSingh1920, 2020-05-26:
Hi Paul,

Thanks for your reply..

I did the changes as suggested by you and added only 5ms delay in loop function but unfortunately on 10ms after giving a few logs the sensor stops reporting the new data. Plz check the attached Pic and logs on 10ms. 

I have Attached the Screenshot and Complete text file log for your reference.

Can you check at your side with the above configuration, if you are able to make it run on 10ms and 5ms delay over SPI Rotation Vector Example and if possible please share logs and screenshot? 

Thanks,
Maneesh

Screenshot => 

[image]

================================================================

Complete Log =>  

[BNO10Ms_5ms Delay_Log.txt](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/files/4679966/BNO10Ms_5ms.Delay_Log.txt)

PaulZC, 2020-05-26:
Hello @ManeeshSingh1920 ,
Looking at your log, it is clear that you are seeing issue #42. The final line:
```Header: 05 00 01 00 Body: 01 Length:5 Channel:Executable```
indicates that the BNO080 has performed a reset because you did not read the SPI data quickly enough.

Here is what I see when running the code on my SparkFun RedBoard (3.3V):

[image]

I have:
- commented out the ```//myIMU.enableDebugging(Serial);```. This disables the debug Serial messages which will take up processor time and slow down the  code
- changed the sensor interval to 5ms: ```myIMU.enableRotationVector(5); //Send data update every 5ms```
- commented out the ```//Serial.println("Doing other things");```
- changed the delay in the main loop to 1ms: ```delay(1); //You can do many other things.```

Here is the complete code:
[Example1-RotationVector.zip](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/files/4680367/Example1-RotationVector.zip)

Please make sure that there is nothing in your main loop that is preventing the code from responding quickly to the BNO080.
I am closing this issue because the library is working correctly and you now have everything you need to solve this issue yourself. But I am going to make it _sticky_ as it is a useful tutorial for other users.
It is possible that it is a problem with the STM32 Nucleo Arduino implementation but that seems very unlikely. I cannot test that - I do not have a Nucleo board.
Best wishes,
Paul