sparkfun/SparkFun_BNO080_Arduino_Library #51: BNO080 not detected

matrangan, 2020-06-01:
### Subject of the issue
I am trying to use a ZIO BNO080 IMU sensor. The ZIO product page points to this library to use. I am unable to use any of the example programs included in this library. No data is ever printed or I get an error that "BNO080 not detected at default I2C address."

### Your workbench
I am using an Arduino UNO R3. [This is the exact IMU](https://www.smart-prototyping.com/Zio-9DOF-IMU-BNO080.html?search=BNO080) I am using. I am also using the qwiic shield from SparkFun to connect the IMU to the arduino. It is being powered through USB. I know the IMUs are not faulty because I have been able to produce data from them using other libraries. However, this is the official library and has much easier examples that I need to use.

### Steps to reproduce
1. Upload any of the example programs
2. Open serial monitor to view output

### Expected behavior
Should execute the program and print respective data

### Actual behavior
No data is printed, or it prints an error "BNO080 not detected at default I2C address."

matrangan, 2020-06-01:
I finally figured out the issue: For some reason, the default I2C address was changed from 0x4B to 0x4A. Adding myIMU.begin(**0x4A**) worked!

closed 2020-06-01 by matrangan

PaulZC, 2020-06-01:
Hi @matrangan ,
Looking at the Zio / Smart Prototyping schematic, they have decided to pull the SA0 pin low by default which sets the address to 0x4A. The [SparkFun board](https://www.sparkfun.com/products/14686) pulls SA0 high by default which sets the address to 0x4B.
On the Zio board, you can change the address to 0x4B by soldering the JP1 split pad.
Best wishes,
Paul

matrangan, 2020-06-01:
Thank you so much for the explanation!