sparkfun/SparkFun_BNO080_Arduino_Library #53: Unclear, license MIT or GPL ?

ldr, 2020-06-20:
Hello!

I would like to use and modify your library, also open sourcing the changes, but before doing this I would like to know what the license is.

In the [LICENSE.md](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/LICENSE.md) file you state it is MIT licensed.

But in the [SparkFun_BNO080_Arduino_Library.h](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/src/SparkFun_BNO080_Arduino_Library.h#L21-L27) and [SparkFun_BNO080_Arduino_Library.cpp](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/src/SparkFun_BNO080_Arduino_Library.cpp#L21-L27) files you refer to the GPL.

And the examples are under even another license [for example example1](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/examples/Example1-RotationVector/Example1-RotationVector.ino#L6) ?

Thanks,

Leon

PaulZC, 2020-06-20:
Hi Leon,
Excellent point! It is unclear...
Please assume MIT - we will update the other references to match this.
Best wishes,
Paul

ldr, 2020-06-20:
Thanks you!

Have a nice day,

Leon

PaulZC, 2020-12-01:
Hi Leon (@ldr),
The license details have now been clarified in v1.1.9.
Closing...
Thank you!
Paul

closed 2020-12-01 by PaulZC