sparkfun/SparkFun_BNO080_Arduino_Library #57: calibartion no sucess allways Unreliable

aortner, 2020-09-22:
### Subject of the issue

sensor allways says
5.94,-3.31,-28.44,Unreliable,	0.00,0.00,-0.00,1.00,High,

after i do every calibration movement..

### Your workbench
blackboard + bno080 i2c connector

* How is everything being powered? usb -c

PaulZC, 2020-09-23:
Hi Andreas (@aortner ),
Can you please check that you are doing _all_ of the movements for the calibration?
Please see [this comment](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/49#issuecomment-630152613) for further details.
Best wsihes,
Paul

aortner, 2020-09-23:
Hi Paul.

Yes i did done all movements. Unreliable dont change..

PaulZC, 2020-09-24:
Hi Andreas (@aortner ),
Example9-Calibrate works OK for me. After performing the calibration movements, I see:
```
-2.50,35.00,34.38,High,	-0.80,0.21,-0.14,0.54,High,
-2.50,35.00,34.38,High,	-0.80,0.21,-0.14,0.54,High,
-2.13,34.19,34.00,High,	-0.80,0.21,-0.14,0.54,High,
```
Your data shows that the magnetometer reading is unreliable. Please make sure you are not close to a strong magnetic field (e.g. from a loudspeaker) or anything containing Iron.
I am going to close this issue as the library is working OK. If you need more help, please ask a question on the IMU forum:
https://forum.sparkfun.com/viewforum.php?f=83
Best wishes,
Paul

closed 2020-09-24 by PaulZC