sparkfun/SparkFun_BNO080_Arduino_Library #58: Adding examples including getReading and multiple access

guihomework, 2020-09-23:
I open this issue here related to a future PR I prepare with an example for the getReadings #55 combined with the multiple access #56,  maybe even with an interrupt if I manage.

This avoids spamming a merged PR

guihomework, 2020-09-23:
Took me some time to debug why I had no interrupt coming, The fact was the interrupt had occurred before I had interrupts enabled, putting the BNO08x in a "data available" state and not producing a falling edge anymore. Enabling interrupts sooner solved the issue.

PR provided #59

guihomework, 2020-11-30:
Dear @PaulZC , I don't need this example of this #59 to me merged in the release-candidate, but would like to know if you had time to look and test the release candidate so that it can be merged in master at some point.
I have been preparing some firmware with the release candidate and ported all my old code to the new "getReadings" method, and all works best so if it was in master, I would be thankful.

best
Guillaume

PaulZC, 2020-12-01:
Hi Guillaume (@guihomework ),
Apologies - updating the BNO080 library has been on my "things to do" list for some time! Other projects got in the way...
I am making some small improvements to the library now, and hope to make a new release in a day or two.
Thank you again for all your support.
Best wishes,
Paul

PaulZC, 2020-12-01:
Hi Guillaume (@guihomework ),
Thank you again for all your contributions. v1.1.9 has just been released.
Closing...
Best wishes,
Paul

closed 2020-12-01 by PaulZC

guihomework, 2020-12-01:
@PaulZC wonderful, this release and new features are a great help for my developments. thanks