sparkfun/SparkFun_BNO080_Arduino_Library #61: No Data from i2c on Nano or Uno

glyndavidson, 2020-10-03:
I'm getting the same results on both Arduino Nano & Uno - No data from the device.

Using i2c on a breadboard (Not Qwiic). 
Pins connected are:
3v3 -> 3v3
GND -> GND
SDA -> SDA
SCL -> SCL

Power light on the breakout is on.

If I enable debugging (myIMU.enableDebugging(Serial)) I get "I2C Timeout".
```
BNO080 Read Example
Accelerometer enabled
Output in form x, y, z, in m/s^2
I2C timeout
I2C timeout
I2C timeout
```

Tried both of these but neither worked:
```
#define I2C_BUFFER_LENGTH BUFFER_LENGTH
#define I2C_BUFFER_LENGTH 32
```

Any suggestions please? Am I missing something really obvious?

PaulZC, 2020-10-03:
Hi @glyndavidson ,
Are you using a SparkFun breakout?
(Could it be this issue? https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/51 )
Best wishes,
Paul

glyndavidson, 2020-10-03:
Hi Paul 

Thanks for getting back to me. 

Yes, I'm using the sparkfun breakout.

Already tried changing the i2c address but that didn't work.

PaulZC, 2020-10-04:
Hi @glyndavidson ,
My next guess is that it is a voltage level problem.
Are both your Arduino Nano and Uno 5V boards?
If they are, they are probably trying to pull SDA and SCL up to 5V, which could produce weird results.
You might find [one of these](https://www.sparkfun.com/products/15439) useful?
Best wishes,
Paul

glyndavidson, 2020-10-04:
I've got an ATMega1284 running on a breadboard at 3.6v. I'll try it with that and see what happens.

PaulZC, 2020-10-16:
Closing... Please re-open if this is still an issue.

closed 2020-10-16 by PaulZC