sparkfun/SparkFun_BNO080_Arduino_Library #63: Latency of gyro rotation vector report?

mark-mcg, 2020-12-04:
Hi all, So I'm trying to use this board as a replacement for head tracking on a VR headset, with the current setup being:
- BNO080 -> Qwik connector to ESP32 WROOM board -> USB serial port connection to an Oculus Quest, with Unity parsing the serial port data. 

However, I'm running into some trouble with the latency of the head tracking. The current setup is:
- baud rate is at 921600, takes about ~2-3ms to write the data to Unity over serial port and parse it (measured RTT of ~4-5ms), and I've verified that there are no delays in Unity with applying the data (applied in `OnPreRender`)
- I2C bus is 400kHz
- `enableGyroIntegratedRotationVector()` called with an update interval of 2ms (500Hz)
- polling `dataAvailable()` in `loop()` and writing the data over the serial port when available

By my estimate, worst case it should take approximately 2ms (serial port) + 2ms (wait for new data) + 1ms (for loop to catch/parse new data) = 5ms. I'm rendering at 72fps, so there is approximately 14ms between frames, which should be ample time to squeeze in one or two updates each frame. However, when I try this, the latency is quite perceivable when I move my head (nauseatingly so), at a guess into the 10s of ms (ie multiple frames worth of delay, but I haven't measured it yet). 

I was just wondering if there's anything I'm likely to have missed in this library for getting the lowest latency rotation vector (other than attempting to enable the board's prediction of the rotation vector)? Thanks!

PaulZC, 2020-12-04:
Hi @mark-mcg ,
I wonder if I2C is the right choice for your application? I wonder if SPI might be a better choice to help reduce the latency?
I would also recommend investigating using the interrupt pin to tell you when new data is available. Guillaume (@guihomework) added [an example](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/examples/Example19-Interrupt/Example19-Interrupt.ino) for this recently.
Best wishes,
Paul

mark-mcg, 2020-12-04:
Thanks for the suggestions @PaulZC , I'll give the interrupt pin a try first, then move on to using SPI instead. Do you have any idea what the ballpark latency difference is like between I2C (with one device on the chain) and SPI? Looks like I'll need to break my soldering iron out!

PaulZC, 2020-12-04:
Hi @mark-mcg ,
Sorry, it's not something I've ever measured...
Good luck!
Paul

PaulZC, 2020-12-29:
Hi @mark-mcg ,
How's it going? Do you need any more help - or can we close this issue now?
Best wishes,
Paul

PaulZC, 2021-01-01:
Closing... Please re-open if this is still an issue.

closed 2021-01-01 by PaulZC