sparkfun/SparkFun_BNO080_Arduino_Library #65: Getting linear acceleration in the NED (North East Down) frame of reference

jerabaul29, 2021-01-13:
Do you know if there is a native way to get the acceleration in the NED (North East Down) frame of reference? If not, I suppose this should be attainable by using the quaternion orientation and the linear accelerations, right? (is the quaternion orientation well describing the orientation relative to NED?).

PaulZC, 2021-01-14:
Hi JR,

You can read the accelerometer data in X/Y/Z - the frame of reference is the chip package. You should be able to combine that with the Euler angles (roll, pitch, yaw) where the frame of reference is deviation from horizontal, yaw is relative to magnetic north. The Euler angles are calculated from the quaternion data by the library. The BNO080 can't provide NED acceleration directly.

I wonder if "Robot Vacuum Cleaner" mode is what you need? It provides Roll, Pitch, Yaw and X/Y/Z acceleration over UART at 100Hz at 115200 Baud. Please have a look at the BNO080 datasheet for more details.

Best wishes,
Paul

closed 2021-01-14 by PaulZC

jerabaul29, 2021-02-17:
FYI: I was able to obtain the quantities in the North-East-Down NED (or rather East-North-Up ENU, see below, but that is the same for all practical matters) frame of reference. The method I used was:

- get the quaternion orientation data (I like better to work with quaternions than Euler angles, more straightforward mathematically, but of course only my taste, Euler angles could certainly be used too :) ). According to the datasheet (reading a bit between the lines, or maybe I did not find the right paragraph in the datasheet), the orientation quaternion produced describes the transformation from IMU to ENU frame of reference.

- get the accelerometer data in the IMU frame of reference

- apply the quaternion rotation to the acceleration to get it in the ENU frame of reference.

All the code is available here: https://github.com/jerabaul29/Artemis_MbedOS_recipes/tree/main/recipes/recipe_IMU_quaternions .

You can find:

- the quaternion utilities (sorry for possibly bad programming style, if you have any feedback let me know, I am not a software engineer at the core :) ): https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/main/recipes/recipe_IMU_quaternions/vector_and_quaternion.h and the .cpp

- an example script: https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/main/recipes/recipe_IMU_quaternions/recipe_IMU_quaternions.ino

I tested with an Artemis RedBoard + Qwiic BNO080 Sf breakout + Qwiic cables. Works fine (module the issues #68 and #69 ).

I have validated that:

- I well get around 9.81 acceleration Up independently of the breakout orientation from this line of output:

https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/b93aaef5978ed483c4ac5b72b01d37c98484e66a/recipes/recipe_IMU_quaternions/recipe_IMU_quaternions.ino#L135

- the direction pointed to by the X-axis of the breakout IMU corresponds with the magnetic reading of my watch, from this line of output:

https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/b93aaef5978ed483c4ac5b72b01d37c98484e66a/recipes/recipe_IMU_quaternions/recipe_IMU_quaternions.ino#L142

I think this may be useful to other users too. Would you consider either writing some small tools that do a similar work, or would you consider a pull request with a few tools that add these functionalities to the library?

If you would consider taking a pull request about that, may you re-open the issue, and may we discuss here some specification / conventions / format stuff? :)

PaulZC, 2021-02-17:
Hi JR,
We are always open to Pull Requests if they add value to the library, provide new features or corrections and - importantly - maintain backward compatibility.
Please include an example that demonstrates how the new features work.
Please also update keywords.txt with the new function names etc..
Best wishes,
Paul

PaulZC, 2021-02-17:
But please target the release_candidate branch as described here:
https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/CONTRIBUTING.md

jerabaul29, 2021-02-17:
Sounds good, will look at it :) .

I may loose internet in the days to come for around 2 weeks, going far into the Arctic, just so that you know why I stop answering if this happens :) . Worst case I will take care of that when I get internet back :) .

PaulZC, 2021-02-17:
Good luck JR - I hope your trip is successful.
I spent 22 years working in a University instrumentation group. I really enjoyed the field trips - they were really rewarding. But always stressful in equal measure too!
All the best,
Paul
PS. I hope the Northern Lights are good!

jerabaul29, 2021-02-17:
Many thanks :) We had very nice Northern lights three days ago. Not seen much of them since, coding 14 hours a day to try to get the prototype instruments to work ^^ (this is why I open so many issues :) ). We have a total of 14 Artemis Global Trackers with us, and we want to turn them into sea ice wave motion, drift, and breakup monitoring devices together with the BNO IMU and some FFT analysis. We have also "well proven" instruments with us (some older designs based on Arduino Mega, RPi, and Vectornav VN100), but having a proof to show to the management that the new, much cheaper and easier to build prototype is promising would be great, that would allow to apply for / plan a large-scale deployment of these motion trackers later on :) .

University instrumentation group must have been really exciting :) . Yes, fieldwork is a bit stressful, but things are still quite OK - we have always made it clear that the stuff I work on now is work in progress :) .