sparkfun/SparkFun_BNO080_Arduino_Library #66: float accuracy values for other quantities than quaternions

jerabaul29, 2021-02-16:
I like very much that the quaternion data can produce a detailed accuracy value in addition to the quality flag:

```cpp
byte quat_accuracy = bno080_imu.getQuatAccuracy();
float quat_radian_accuracry = bno080_imu.getQuatRadianAccuracy();
```

I only find documented the quality flag, not detailed float value, for the other components (acceleration, gyro, magnetometer). Any way to access detailed float value uncertainties on these too / if not, is this something that can be added, or is it just not produced by the IMU?

PaulZC, 2021-02-17:
Hi JR,
Please see my reply in #67. If the data sheet suggests such accuracy figures are available, and which the library does not currently make available, please raise a “Feature Request “ issue and we will look at it as soon as we can.
Best wishes,
Paul

jerabaul29, 2021-02-17:
I agree with you, I cannot find much. I wonder if it is possible that the datasheet is not very complete.

- there is some mention of some Q-point and accuracy in paragraph 1.4.4, but I must say I do not really know what this means, no idea if this may be what we are looking for.

- the thing that was (positively) surprising to me is the accuracy value for the Quaternion output that has a numeric component, as visible in https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/314ba5d9d9cfa4a760e0e86267f26ce54858f7a2/examples/Example1-RotationVector/Example1-RotationVector.ino#L70 . I do not know how @nseidle got to know that this data was here though, and where its format was defined...

The thing that makes me think such data may be available is that:

- the internal firmware must run some form of Kalman filter or similar fusion algorithm which must have some internal way of representing uncertainties to actually work

- these filters usually have both some prediction and correction steps, and the mismatch between the 2 is also related to both the uncertainty in the state (orientation, acceleration, etc) of the IMU, and the noise of the sensors

- so there may be some good estimates of the uncertainties available somehow inside the processing.

No idea if / how this could be accessed though, but probably the manufacturer of the firmware could say something more (?). I just got good hopes due to the Quaternion uncertainty data being available.

PaulZC, 2021-02-18:
Hi JR,
Looking at the "SH-2 Reference Manual", I only see **true** "Accuracy Estimates" for:
- Rotation Vector (0x05)
- Geomagnetic Rotation Vector (0x09)
- ARVR-Stabilized Rotation Vector (0x28)

and we do already extract ```rawQuatRadianAccuracy``` from all three.

The "accuracy" or "status" byte for the accel, gyro and mag is:

[image]

There does not appear to be an equivalent 16-bit value for those.

Sorry - that's the best we can do!
Paul

closed 2021-02-18 by PaulZC

jerabaul29, 2021-02-18:
Ok, thanks :) .

Maybe I could take contact with Hillcrest Labs directly and ask them if there is more than what is in the datasheet. Do you have a contact address that you would recommend that I use? I could put you in copy if you want :) .