sparkfun/SparkFun_BNO080_Arduino_Library #67: low power

jerabaul29, 2021-02-16:
I would like to use this IMU for some low power applications.

- is there any way to get the IMU to reduce power use / enter a sleep mode by sending some I2C command? Can it be woken up using an I2C command too?

- if not / in addition, do you have any documentation / could you add some tutorial and documentation about how the Wake / WAK pin? How should it be used?

PaulZC, 2021-02-17:
Hi JR,
All I can suggest is that you have a read through the BNO080 data sheet. If there is a “low power” feature in there, which the library does not currently support, then please raise a “Feature Request” issue and we will look at implementing it as soon as we can.
We will always try our best to support requests for new features in the library, resources permitting.
Best wishes,
Paul

jerabaul29, 2021-02-17:
That sounds good :) .

The main "traces of information" I can find in the datasheet (I have a copy of the BNO080 datasheet v1.3 with me, internet is limited, cannot do a more up to date search) are:

- "Includes power management to optimize power" on the first page
- "Idle Power" in Figure 6-15 that should go down depending on which rail is used to power to 0.12 or 0.01 mA. I think this is the mode I would be most interested in being able to easily enable / disable.

But I am not sure of how this "idle power" sensor mode can be activated. I cannot find more information in the datasheet. Wonder if the manufacturer could provide more information?

This power mode would be very nice to me. If it can be switched on and off through I2C (or is it naive? does this idle power mode correspond to I2C off you think?), may that also be a way to go around the I2C issue, by switching on the I2C communication "at the very start" and keeping it always on, just in idle mode? I am by no ways an expert on this though, so this may be very wrong...

PaulZC, 2021-02-18:
```modeSleep``` is included in v1.1.10.
Closing...

closed 2021-02-18 by PaulZC

PaulZC, 2021-03-08:
Please see [this update](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/72#issuecomment-792985261) in #72 . The [updated example](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/release-candidate/examples/Example20-Sleep/Example20-Sleep.ino) in the release_candidate branch now draws a sensible amount of current while asleep.