sparkfun/SparkFun_BNO080_Arduino_Library #71: Feature request: idle power / low power mode

jerabaul29, 2021-02-17:
First, not sure if this is how feature requests should be opened, sorry if this is wrong.

The datasheet, Fig. 6-15 mentions that in "idle power", the sensor can use "0.12 mA (VDDIO Rail)" or "0.01 mA (VDD Rail)". I cannot find more about this in the datasheet.

- Any idea how this works? Maybe the manufacturer has more information?
- Is it possible that this idle mode could be switched on / off from the I2C and / or other protocols? If so, could this be added?

jerabaul29, 2021-02-17:
Related to #67 . Feel free to close the issue if you accept this feature request.

jerabaul29, 2021-02-17:
No idea where the author found this, and I would need to test (but I do not have good power measuring equipment; anyone who has?), but:

https://github.com/jps2000/BNO080/blob/c5a7a00373e6cdc1503349e019face416b78b79c/Arduino_BNO080.ino#L554-L559

jerabaul29, 2021-02-17:
(copy for archive:

```cpp
// This code resets the BNO080 or put it into sleep
void reset_BNO(){
uint8_t action = 2;                  // 1 = reset, 2 = on; 3 = sleep
uint8_t reset_BNO[4] = {0,1,0,action};
I2c.write(BNO_ADDRESS,5,reset_BNO,4);
```

PaulZC, 2021-02-18:
Hi JR,

OK. Found it. You can use an "executable channel" command to put the BNO080 into "sleep" state. "On" wakes it again.

[image]

For speed, I'll add two new commands, on_BNO and sleep_BNO, based on reset_BNO (which also uses the executable channel).

Watch this space,
Paul

jerabaul29, 2021-02-18:
Many thanks, that would be amazing, cannot thank you enough :)

I suspect this will also "solve" the I2C connection issue: I can just "get it to work" once at the very beginning, if necessary going through several booting up. Then I will never really switch off the sensor, only put it to sleep and wake it up :) ,

jerabaul29, 2021-02-18:
Arrgh, regarding the reset issue on I2C: no, I am just forgetting that I would  probably need to switch off the Artemis for low power, which will shut off the I2C from the Artemis side I guess... Another option is that I do not really switch off the Artemis - I do not need that low power, if the Artemis draws 1mA instead of 0.015 mA it is good enough to me. Will look into this :) .

PaulZC, 2021-02-18:
Partial success...
```modeSleep``` works, but only reduces the current draw by about 7mA. It does not get the BNO080 back to its initial low power state. I'm not sure what's preventing that as we don't seem to be using any of the "wake" features. ```softReset``` doesn't seem to do the trick either.
That's as much as I can do for now...
The new code and example are [here](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/pull/73/files)
Sorry this isn't a complete solution for you,,
Paul

closed 2021-02-18 by PaulZC

jerabaul29, 2021-02-18:
Ok, many thanks for your help! Yes, 7mA is a bit high. Maybe I should / could take contact with Hillcrest labs? Similar to the other issue, do you have an address you recommend asking them on / should I put you in copy?

jerabaul29, 2021-02-18:
Could the 7mA be in part related to the LED and the ? I am a bit confused, looked on the schematics here https://raw.githubusercontent.com/sparkfun/Qwiic_IMU_BNO080/master/Hardware/Qwiic_IMU_Bonobo.pdf but could not find the LED on it (probably I am just missing something).

jerabaul29, 2021-02-18:
(actually there are quite a few resistors on the board that I do not find on the schematics; probably just tired.

jerabaul29, 2021-02-18:
Sorry, long day, and I read too fast: "by" 7mA, not "to" 7mA, right? I will check on my side tomorrow.

PaulZC, 2021-02-19:
Hi JR,
Just to clarify:
When the BNO080 board is first powered on, before the software starts, it draws around 2mA. Most of this is the LED.
When the sensor is running at full speed, it draws around 15mA.
With ```modeSleep```, the current drops **by** 7mA to around 9mA. I couldn't get it to go any lower. A ```softReset``` didn't help.
So, at the moment, I don't know how to get it back into its original low power state except by power-cycling it.
The datasheet divides the current draw into "VDD Rail" and "VDDIO Rail". The draw from both is around 7mA when the chip is running. My guess is that sleep disables whatever is drawing current from one, but not both.
You need the [Qwiic Power Switch](https://www.sparkfun.com/products/16740) but sadly it is out of stock at the moment.
Best wishes,
Paul

jerabaul29, 2021-02-19:
Ok, thanks for your help, this is very helpful :) .

jerabaul29, 2021-02-19:
PS: do you think this is an issue with the breakout or the BNO (and in this case, should the manufacturer be notified? ...).

jerabaul29, 2021-02-19:
(also, curious if some of the issues we have here are fixed in the BNO085; if I remember well, the manufacturer fixed quite a few hardware issues on the 85 upgrade, right?).