sparkfun/SparkFun_BNO080_Arduino_Library #72: discussing about the I2C hanging issues

jerabaul29, 2021-02-17:
This is moslty just for discussing / brainstorming.

There is this really annoying issue with the BNO080 hanging when put on I2C to the Artemis.

Any feedback you received about connecting the BNO080 through UART or SPI instead? Does that work without hanging? Of course I would prefer I2C, but if needed, I can use something else.

It is very late, will try tomorrow, but would be great to know if there are some known issues there too / if investigating this would waste my time. In this case, I would stick to I2C and the hard reset solution ^^ .

jerabaul29, 2021-02-18:
I closed issues #68 and #69 in favor of this discussion.

I spent quite a bit of time investigating these issues, so I thought I would give a small update here. That may be useful for other users who hit the same issues.

Because of some issues with querying several data feeds with the current SF code (see #70 ), I use the Adafruit code. There is a bit of a hack to get it to work on the Artemis board, see the recipe

jerabaul29, 2021-02-18:
@PaulZC I was thinking more about that. Any way that we could restart the I2C port, i.e. power it up and down, from the watchdog, without the need to restart the whole board? That may be a workaround to reset the port without needing the data in RAM, right? Any idea this could be doable / is there a HAL call for that?

jerabaul29, 2021-02-18:
I am trying to reset the I2C hardware from within the interrupt with something like this:

```cpp

volatile bool force_wait {true};  // no delay() in interrupt; force reading and writing a volatile variable
volatile int remaining_watchdog_tries {10};

// Interrupt handler for the watchdog
extern "C" void am_watchdog_isr(void)
{
  wdt.clear();
  
  if (remaining_watchdog_tries > 0){
    wdt.restart();
    remaining_watchdog_tries -= 1;
    Serial.print(F("retry I2C "));
    
      Wire.end();
      am_hal_pwrctrl_periph_disable(AM_HAL_PWRCTRL_PERIPH_IOM0);

      // hand-made delay, to make sure power has time to go down
      for (int i=0; i<1000000; i++){
        force_wait = !force_wait;
      }
      
      am_hal_pwrctrl_periph_enable(AM_HAL_PWRCTRL_PERIPH_IOM0);
      Wire.begin();
    
  }
  else{
    // reset
    while (true){;}
  }
}
```

jerabaul29, 2021-02-18:
I even tried to disable and enable all the periphs (IOM0 to end), but that does not help.

I do not know / have no way to know as far as I know (some method to get this information?) where the code hangs. I would say that now in around 90% of the cases I manage with the looping to get contact with the BNO, and in around 1 in 10, the code hangs when trying to begin the BNO, the watchdog kicks in, tries 10 times to restart the I2C, but with no success, leading to a full reset.

If there was a way to trigger a reset of the right I2C stuff from the watchdog, rather than by doing a full chip reset, that would practically solve the problem. I do not have the JTAG or similar debugging tool nor the experience to debug down to this level / get this to work. If you have any advices @PaulZC that would be amazing.

PaulZC, 2021-02-18:
Hi JR,
Like I said in [this comment](https://github.com/sparkfun/OpenLog_Artemis/issues/67#issuecomment-771473645), I have already spent _hours_ looking at this. But, I must admit, I haven't tried JTAG. That might be a way forward - but I really don't have time to look at it right now.
If you want to dig into it, start looking in here:
https://github.com/sparkfun/Arduino_Apollo3/blob/v1/cores/arduino/ard_sup/iomaster/ap3_iomaster.c#L759
That function is used for both I2C and SPI - not just SPI.
If I recall correctly, the behavior is that the hardware gets into a state where ```am_hal_flash_delay_status_check``` always times out, preventing any further I2C transactions and causing the bus to hang.
If you solve this, I definitely owe you a beer!
Paul

jerabaul29, 2021-02-19:
@PaulZC just FYI: it looks like I "empirically" found a recipe that works all the time at starting the BNO with Artemis, at least on the Artemis RedBoard and AGT I have here for testing, and with the couple of BNOs I have here too ("works on my computer" symptom).

This recipe:

https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/main/recipes/recipe_IMU_quaternions/recipe_IMU_quaternions.ino

With in particular this snippet:

https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/5984c4c799a52c4979392e0e71386a56da78fb3a/recipes/recipe_IMU_quaternions/recipe_IMU_quaternions.ino#L36-L67

Has been working without any issues now for me several tens of times. Other similar snippets with different timing or checks fail regularly, but just this one seems to work.

I would be interested to hear if other users are able to use this snippet to get a reliable startup.

jerabaul29, 2021-02-19:
(you really do not want to know how many hours I spent having trial and errors to get to this one...).

PaulZC, 2021-02-19:
Hi JR,
Sorry, I should have mentioned that earlier. I did notice that calling ```Wire.setClock(400000);``` _before_ attempting to start the BNO improved the performance. Why should 400k be more reliable than 100k? I really have no idea. In my mind it just confirms that there is something 'marginal' about the combination of the Artemis hardware with the BNO.
Best wishes,
Paul

jerabaul29, 2021-02-19:
Yes, there is something really fishy. I have been "test trying" for around 1 hours to get this snippet to crash in the same way other similar snippets do, "luckily without" success. I think there is a broken piece of code somewhere and that somehow having just this combination of delays allows to avoid it.

jerabaul29, 2021-02-22:
For the records, Teensy forum has a loooong discussion around that:

https://forum.pjrc.com/threads/58268-Sparkfun-BNO080-Teensy-3-6-gt-gt-4-0

Read through it now, and not completely sure what the conclusion was ^^ .

adamgarbo, 2021-02-25:
Hi folks,

I just received my SparkFun BNO080 boards yesterday and spent a couple of hours trying to troubleshoot why they wouldn't initialize with Artemis-based boards until I found these issues on GitHub. 

I'm still catching up, but it sounds like the main takeaway is that the Artemis and BNO080 are, for the moment, incompatible over I2C?

Thanks, @jerabaul29 and @PaulZC for your work in looking into this.

Cheers,
Adam

jerabaul29, 2021-02-25:
Hi @adamgarbo ,

Yes, had the same kind of troubles unfortunately :( .

A few notes:

- with a snippet in the kind of the previous one (actually a bit longer, see below), I was having a good success rate at waking up the BNO and connecting to it. Together with a watchdog reset it is quite ok to use. The only problem is the loss of RAM data at reset, but this could be solved by using the "EEPROM" stuff for holding important data I suppose :) . I have 2 intruments in the middle of the Arctic using this solution, together with the quaternion analysis similar to what is done in https://github.com/jerabaul29/Artemis_MbedOS_recipes/tree/main/recipes/recipe_IMU_quaternions , and they seem to work nicely.

- I found out that the RVC (Robot Vacuum Cleaner) mode, using UART at baud 115200, works perfectly and is very reliable. I have a recipe here (it was on the Artemis Global Tracker, so I had to trick a bit and use a hardware serial to talk to the BNO as it is quite fast, and reconfigure the serial to the Iridium as a software serial since the hardware serial got used, but this is fine, as software serial works very well at the baudrate 9600 of the Iridium modem). See: https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/main/recipes/uart_rvc_bno08x/uart_rvc_bno08x.ino . I have 12 instruments in the middle of the Arctic using this solution, and it seems to work perfectly.

So my advice would be, if it is ok for your application, to use the RVC mode. As far as I found out, the only negative points for my use were that:

- it outputs Euler angles and not quaternion, which is much less satisfactory from a mathematical point of view, and also, seem to have a gimbal lock when tilt is around 90 degrees. For my (and possibly your I guess :) ) applications, this is not a problem though.

- output is fixed at 100Hz. This is fine for me though.

- the orientation is given relative to the direction at startup, and not relative to the North, and I do not think that there is a way to get information about heading relative to the North. For my current application this is not an issue, but for later application, this will  become a big issue from my side. :(

So it is in general quite ok to use, but it will make for a good story in my department, when I explain to my boss and colleagues that our instruments believe that they are robot cleaners ^^ .
 
These problems are very sad, because this is an amazing sensor. I would be very curious @PaulZC to know if using the BNO085 could fix this issue, and if Sparkfun would be ok shifting to these in the future :) .

PS: my current snippet for the I2C. Ugly, ad-hoc, hacky, but seems to do the trick around 90% of the time in my tests:

```cpp

bool BNO080_Manager::_power_start(void){
  turn_bno_off();
  delay(1000);

  Wire1.begin();
  delay(50);
  Wire1.setClock(400000);
  delay(50);
  wdt.restart();

  delay(2000);
  wdt.restart();
  turn_bno_on();
  delay(5000);
  wdt.restart();

  bno080_imu.begin(BNO080_DEFAULT_ADDRESS, Wire1);
  delay(50);
  bno080_imu.enableDebugging(Serial);
  delay(50);
  wdt.restart();

  bno080_imu.enableRotationVector(imu_output_period);
  // bno080_imu.enableAccelerometer(imu_output_period);
  wdt.restart();

  delay(5000);
  wdt.restart();

  while (!bno080_imu.begin(BNO080_DEFAULT_ADDRESS, Wire1)){
    Serial.println(F("try to begin"));
    delay(100);
  }

  delay(5000);
  bno080_imu.begin(BNO080_DEFAULT_ADDRESS, Wire1);

  wdt.restart();

  Serial.println(F("bno set up, start measuring"));
  wdt.restart();
  
  return true;
}
[... 89 lines cut ...]

    if (_first_measurement_start()){
      start_succes = true;
      break;
    }
    else{
      stop();
      delay(1000);
    }
  }

  if (start_succes){
    Serial.println(F("success starting bno"));
  }
  else{
    Serial.println(F("no success starting bno"));
    stop();

    Serial.println(F("we reboot"));
    while(true){;}
  }
  
  return start_succes;
}
```

adamgarbo, 2021-02-25:
Hi folks,

I tested a large number of BNO080 examples using v2.0.5 of the Apollo3 Core and a SparkFun Redboard Artemis/Redboard Artemis Nano. I would have also included theMicroMod Artemis, but v2.0.5 doesn't have this board definition yet. 

I am happy to report that I was not able to reproduce any of the I2C hangs or sensor initialization failures with the examples I tested. Perhaps this bug is specific only to v1.2.x? 

@jerabaul29, with SparkFun announcing the EOL of v1.x of the Apollo3 Core yesterday, I believe it makes sense for us to focus our efforts on v2.x. User contributions will be a big help in getting it to where it needs to be.

Cheers,
Adam

jerabaul29, 2021-02-26:
Ok, interesting. Yes, there are reports of issues with the BNO080 all over the internet with a variety of boards. For my part, very curious to know if the BNO085 fixes these issues.

jerabaul29, 2021-02-26:
@PaulZC @nseidle I am planning to build a new batch of between 5 to 10 instruments based on Artemis Global Tracker and some variation of BNO IMU in the course of the next few months. Any hope you can check if upgrading to the BNO085 can offer a fix to this problem?

PaulZC, 2021-02-27:
Hi JR,
Unfortunately we don't have a product based on the BNO085 and, to the best of my knowledge, we don't plan to release one any time soon. (Fundamentally, I don't have a BNO085 I can test.)
We have just added support for the Digital Motion Processor on the [ICM-20948](https://www.sparkfun.com/products/15335). You can now read 6-axis or 9-axis Quaternion data from that chip and it works well with the Artemis on I2C. The release notes are [here](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.0).
You can put it into a low power state but, as @adamgarbo [observed](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/21), the sleep current on our breakout is not as low as it could be. This has to be due to the level-shifting circuitry as we use the same sensor on [OpenLog Artemis](https://www.sparkfun.com/products/16832) and can get the sleep current there below 100uA.
Best wishes,
Paul

jerabaul29, 2021-02-27:
Thanks for the detailed explanations :) .

If I understand right, the BNO085 is virtually exactly the same chip as the BNO080, except from a few fixes. It looks like for example Adafruit just swapped the BNO080 for a BNO085 in production, using the exact same breakout board and libraries. I will definitely buy one of their BNO085 to test a bit and see if it helps. The fact that these 2 chips seems to be very easily swappable, with the BNO085 being a drop-out backwards-compatible replacement, is what would make me very interested in seeing Sparkfun offering it.

jerabaul29, 2021-02-27:
Another thing is that I am interested a bit in the power consumption, but most important for me is accuracy. It seems that, on this point, the BNO08x is the best, right? Or do you know @PaulZC  of an accelerometer with even better performance?

That starts to drift a bit from the original discussion, but if you have recommendation about what would be the "best possible accelerometer under at or under 50 to 100USD" in the sense of accuracy, I would be very interested; may be possible to have both an high accuracy accelerometer and an IMU to get acceleration and orientation on the same board for my part, i.e. would not be a worry to have 2 chips with a bit of overlap, if that would be the way to get best accuracy.

PaulZC, 2021-02-28:
Hi JR,
Indeed, we are going off-topic but let's keep talking...
Have you got a feel for what you need in terms of: Gyro Angular Random Walk, Axis to Axis Alignment, Gyro In-Run Bias Stability, Accel In-Run Stability, Factory Calibration etc.?
This parameter table from Analog Devices might help:
https://www.analog.com/en/parametricsearch/11172#/
Cheers,
Paul

jerabaul29, 2021-02-28:
Thanks for the link :) .

I think this will become much broader discussion, would you @PaulZC @adamgarbo be ok to continue discussing here? https://forum.sparkfun.com/viewtopic.php?f=139&t=55182&p=223524#p223524

PaulZC, 2021-03-08:
Hi JR (@jerabaul29 ),
I hope all is going OK for you.
I _think_ I might have cured this, or improved it at least?
Can you please pull the [release_candidate](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/tree/release-candidate) branch and give it a try?
The BNO080 seems to start cleanly on Artemis using I2C whereas it used to fall over regularly (as you know). I'd be very interested to see if you see the same improvement?
Thanks!
Paul

jerabaul29, 2021-03-08:
Hi Paul,

That sounds amazing, many thanks :) .

I am exhausted after the fieldwork of last month and taking a bit of holidays. @adamgarbo do you think you could have a look at this? :) .

adamgarbo, 2021-03-08:
Morning,

I just tested the BNO080 with a SparkFun Redboard Artemis on v2.0.5. It appears that if the sensor and/or system is power cycled, the initialization failures will still manifest themselves.

```
09:17:52.631 -> BNO080 Read Example
09:17:53.093 -> Rotation vector enabled
09:17:53.093 -> Output in form roll, pitch, yaw
09:17:53.336 -> 3.7,-0.8,-13.9
09:17:53.407 -> 3.7,-0.7,-13.9
09:17:53.440 -> 3.8,-0.7,-13.9
09:17:53.477 -> 3.9,-0.8,-13.9
09:17:53.546 -> 3.9,-0.7,-13.9
09:17:53.581 -> 4.0,-0.7,-13.9
09:17:53.648 -> 4.0,-0.7,-13.9
09:17:53.686 -> 3.9,-0.7,-13.9
09:17:53.757 -> 4.0,-0.6,-13.9
09:17:53.791 -> 3.9,-0.7,-14.0
09:17:53.828 -> 4.1,-0.7,-13.9
09:17:53.903 -> 4.7,-0.6,-14.0
09:17:53.938 -> 5.6,-0.1,-13.5
09:17:53.975 -> 6.0,0.3,-12.9
09:17:54.043 -> 6.1,0.5,-12.6
09:17:54.078 -> 6.8,0.9,-12.8
09:17:54.154 -> 6.5,0.9,-12.9
09:17:54.191 -> 6.9,1.1,-12.7
09:17:54.225 -> 7.0,1.1,-12.7
09:17:54.295 -> 6.9,0.9,-12.8
09:17:56.648 -> BNO080 Read Example
09:17:57.108 -> BNO080 not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...
```

I also tried testing the Qwiic Power Switch with the Redboard Artemis but I'm not sure if this combination is compatible. I can observe the softReset forcing the QPS to shut down (LED turns off)? I haven't looked at the changes to the library yet, but is this causing a hard reset to the I2C bus?

```
09:21:00.470 ->  BNO080 Read Example
09:21:07.061 -> softReset: first receivePacket
09:21:07.127 -> softReset: second receivePacket
09:21:07.127 -> softReset: complete
09:21:07.127 -> SW Version Major: 0x3 SW Version Minor: 0x2 SW Part Number: 0x98A498 SW Build Number: 0x172 SW Version Patch: 0x7
09:21:18.263 -> softReset: first receivePacket
09:21:18.335 -> softReset: second receivePacket
09:21:18.335 -> softReset: complete
09:21:18.335 -> SW Version Major: 0x3 SW Version Minor: 0x2 SW Part Number: 0x98A498 SW Build Number: 0x172 SW Version Patch: 0x7
```

Cheers,
Adam

PaulZC, 2021-03-08:
Hi Adam,
Thanks for looking at this.
The softReset has always been there - it is part of the .begin functionality - but the debug messages are new. It should have no effect on the Qwiic Power Switch. The I2C address is different - but only two bits different for the default address on the SF Breakout (0x41 vs 0x4B).
I've tried hooking up a QPS here and I don't see the LED turn off. Sounds like something very strange is happening? How are your pull-ups configured?
Also, I've been meaning to ask: how long do you wait, after turning the QPS on, before you call .begin?
Thanks!
Paul

adamgarbo, 2021-03-08:
Hi Paul,

There is a 1 second after enabling the QPS and initializing the BNO080. QPS "IN" pullups enabled, "OUT" disabled. I just ran another series of tests with the SparkFun Redboard Artemis and QPS and had better results. The QPS was able to repeatably power cycle the BNO080. I'll run a longer test to see how it performs.

I did encounter a strange Mbed OS error on more than one occasion (shown below), but I don't know if it is related. It occurs when powering up the Redboard Artemis with the QPS/BNO080 attached or when uploading new code.

With the BNO080 connected directly to the Redboard Artemis, it does appear to be more stable, though pressing the "RESET" button will still produce the usual I2C initialization errors:

```
11:29:49.092 ->  BNO080 Read Example
11:29:49.542 -> BNO080 not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...
```

Cheers,
Adam

```
11:20:54.802 ->  BNO080 Read Example
11:20:55.295 -> 
11:20:55.295 -> ++ MbedOS Fault Handler ++
11:20:55.295 -> 
11:20:55.295 -> FaultType: HardFault
11:20:55.295 -> 
11:20:55.295 -> Context:
11:20:55.295 -> R0: 0
11:20:55.295 -> R1: 100070AB
11:20:55.295 -> R2: 1
11:20:55.295 -> R3: A3009150
11:20:55.295 -> R4: 100070AB
11:20:55.295 -> R5: 0
11:20:55.295 -> R6: 4
11:20:55.295 -> R7: 4
11:20:55.295 -> R8: 0
11:20:55.295 -> R9: 0
11:20:55.295 -> R10: 0
11:20:55.295 -> R11: 0
11:20:55.295 -> R12: 25231
11:20:55.295 -> SP   : 10007078
11:20:55.329 -> LR   : 125D9
11:20:55.329 -> PC   : 125E0
11:20:55.329 -> xPSR : 21000000
11:20:55.329 -> PSP  : 10007010
11:20:55.329 -> MSP  : 1005FF70
11:20:55.329 -> CPUID: 410FC241
11:20:55.329 -> HFSR : 40000000
11:20:55.329 -> MMFSR: 0
11:20:55.329 -> BFSR : 82
11:20:55.329 -> UFSR : 0
11:20:55.329 -> DFSR : 0
11:20:55.329 -> AFSR : 0
11:20:55.329 -> BFAR : A3009154
11:20:55.329 -> Mode : Thread
11:20:55.329 -> Priv : Privileged
11:20:55.329 -> Stack: PSP
11:20:55.329 -> 
11:20:55.329 -> -- MbedOS Fault Handler --
11:20:55.329 -> 
11:20:55.329 -> 
11:20:55.329 -> 
11:20:55.329 -> ++ MbedOS Error Info ++
11:20:55.329 -> Error Status: 0x80FF013D Code: 317 Module: 255
11:20:55.329 -> Error Message: Fault exception
11:20:55.329 -> Location: 0x125E0
11:20:55.329 -> Error Value: 0x100060C8
11:20:55.363 -> Current Thread: main Id: 0x100044B8 Entry: 0x25005 StackSize: 0x1000 StackMem: 0x10006128 SP: 0x10007078 
11:20:55.363 -> For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=SFE_ARTEMIS
11:20:55.363 -> -- MbedOS Error Info --
```

adamgarbo, 2021-03-08:
I forgot to mention that the LED on the QPS turning off was due to a coding error on my part.

PaulZC, 2021-03-08:
Well, it's progress. Kind of... ;-)
I do see occasions where: if I reset the Artemis by re-starting the Serial Monitor, the SDA line sticks low, permanently. And then (of course) trying to .begin the BNO fails. I suspect the BNO is pulling it low. Maybe because it was interrupted part way through a data transfer ?
I'm wondering if the BNO would behave better if we disabled the data stream or put it into ```modeSleep``` before turning the power off?

adamgarbo, 2021-03-08:
I think the BNO080 is just too darn smart for its own good! 

Can we also pause for a moment to appreciate the irony that the BNO080 is based on the Cortex-M0+ but throws a fit when we want it to work with other Cortex-M0+ or M4 processors?

PaulZC, 2021-03-08:
"Interesting" thing: if you see the "SW Version Major" debug message then the BNO has begun successfully. You only see that if the PRODUCT_ID_RESPONSE is correct. But, yes, it then doesn't generate any data. Why does it sometimes ignore the ```setFeatureCommand``` which enables the sensor report? I think that's what I need to figure out next...

adamgarbo, 2021-03-08:
Paul Stoffegren fixed the same issue we're having with the Teensy 4.0 (Cortex M7) and the BNO080 last year, though I'm not exactly sure what was fixed and if it's applicable to the M4 or M0+-based processors:

https://forum.pjrc.com/threads/58268-Sparkfun-BNO080-Teensy-3-6-gt-gt-4-0?p=226478&viewfull=1#post226478

Would SparkFun be eligible for support directly from CEVA/Hillcrest Labs?

PaulZC, 2021-03-08:
Yeah. I'm pretty sure that's it. I've restructured the [sleep example](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/release-candidate/examples/Example20-Sleep/Example20-Sleep.ino) and it now seems to start reliably too. I was seeing the same thing as you. The begin was successful but then the ```setFeatureCommand``` failed periodically. Please kick the tyres if you can spare the time! Thanks!
The good news (for you and JR (@jerabaul29 )) is that the current draw in sleep is now back down to the ~2mA LED background current. The previous version continued to pound the I2C bus while the BNO was asleep, raising the current draw by several mA!

Re: CEVA. Yes, but they're busy guys. We try not to bother them unless there's something really major we need to ask them about. (Like the SPI bus crash feature - if you don't read the data quickly enough). I guess this issue is getting up there though...!

All the best,
Paul

PaulZC, 2021-03-08:
I’m still using 1.2.1. Haven’t been brave enough to upgrade yet. But that time is coming soon!
All the best,
Paul

adamgarbo, 2021-03-08:
Interesting! I found that when I downgraded to v1.2.1 and conducted the same tests, the sensor was significantly less stable than on v2.0.5. 

What instrument are you using to monitor the I2C data? It might be something worth investing in.

Cheers,
Adam

PaulZC, 2021-03-08:
https://www.sparkfun.com/products/15033

jerabaul29, 2021-03-27:
Just picking things up: what is the current status on this? :) . @adamgarbo are you using the BNO in 'production' and does it work fine, or did you give up and go for another IMU?

adamgarbo, 2021-03-27:
Hi @jerabaul29,

Unfortunately, I haven't had time to troubleshoot the BNO080 further. With how unreliable it's been, I think it'll be too risky to include in the instrumentation I'll be deploying this year, especially considering it can crash the entire I2C bus.

I'm currently exploring alternative IMU sensor options. Jim Remington has written a library for the LSM9DS1 that enables tilt-compensated compass measurements and sensor fusion (https://github.com/jremington/LSM9DS1-AHRS). I've been waiting for SparkFun to get the sensor back in stock (https://www.sparkfun.com/products/13944).

Please let me know if you have any suggestions.

jerabaul29, 2021-03-27:
Ok, thanks for the update!

I just ordered / will be looking at replacing it with the ICM-20948 .

adamgarbo, 2021-03-27:
Fair warning that the ICM-20948 is a black box and has its own issues. Paul has done lots of great work with the library, but I haven't been able to get reliable measurements from it yet. It also has a high quiescent draw of ~4 mA due to the switching circuitry.

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/40

jerabaul29, 2021-03-27:
Ok, thanks for the warning. This is quite annoying - there are so many IMUs that 'should be a perfect fit' but looks like none of them is truly working flawlessly :( .

PaulZC, 2021-03-28:
Hi JR,
The release_candidate branch contains some changes which helped the BNO080 start correctly on Artemis / Apollo3 using v1.2.1 of the core. The key thing was inserting an extra delay after the soft reset. If you try to talk to the BNO080 too soon, it returns an invalid or corrupt packet where the length is huge. That was confusing the library and causing it to try to read far too many bytes from the BNO. The extra delay cures this. But the changes I've made seem to cause extra problems on the PORTENTA. I don't have PORTENTA hardware so I can't investigate further...
All the best,
Paul

jerabaul29, 2021-03-28:
That sounds great @PaulZC :) . So do you mean this new fix on dev branch is working 100% of the time? @adamgarbo you were naming some issues even with the newer fixes, does that apply to this dev branch code?

jerabaul29, 2021-03-28:
To continue this discussion @PaulZC , from my 'small egoistic' perspective I am only interested in using the BNO with the Artemis platform, but I really need high reliability, so Portenta is not a big deal for me :) Would even say that if there were an 'artemis tuned' branch that would work perfectly with specifically the Artemis, that would be excellent as far as I am concerned :)

PaulZC, 2021-03-28:
The Portenta issue was - I think - a silly debug print mistake which I've just corrected:
https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/80#issuecomment-808867019
Please give release_candidate a go when you have time. Thanks!

jerabaul29, 2021-03-28:
Sounds good, I want do some detailed testing of the updated branch, hope to have a bit of time for that next week :) .

adamgarbo, 2021-03-28:
Hi @jerabaul29,

I quickly tested the release-candidate branch with the Artemis on v1.2.1 and it appears to still have issues with initialization and power cycling (reset button).

I purchased a logic analyzer from SparkFun but don't have the time to work on this at the moment. Please let me know what you find!

```
09:02:23.510 -> BNO080 Read Example
09:02:24.004 -> BNO080 not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...
09:02:29.106 -> 
09:02:29.106 -> BNO080 Read Example
09:02:29.524 -> BNO080 not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...
09:02:32.671 -> 
```

Cheers,
Adam

jerabaul29, 2021-03-29:
Ok, thanks, sad to hear that, will do some tests later this week (I hope, busy week).

adamgarbo, 2021-03-29:
Hi @jerabaul29,

I have a test running with the BNO080 and MicroMod Artemis Processor + Data Logging Carrier Board. I am power cycling the sensor 5 seconds ON and 5 seconds OFF, reinitializing each time. 

So far, it does appear to be more stable, but the important thing was to remove the `while (1);` in the example if the initialization fails. I've read elsewhere that the failure to initialize should be ignored and the sensor can be read from regardless. 

When you enable debugging, you can observe there is a timeout, but the sensor does eventually initialize properly:
```
10:33:21.547 -> sendPacket(I2C): endTransmission returned: 4
10:33:21.690 -> softReset: first receivePacket
10:33:21.796 -> waitForI2C: I2C timeout when expecting 4 bytes. 0 were available
10:33:21.866 -> softReset: second receivePacket
10:33:21.866 -> softReset: complete
10:33:21.866 -> SW Version Major: 0x3 SW Version Minor: 0x2 SW Part Number: 0x98A498 SW Build Number: 0x172 SW Version Patch: 0x7
```

When I upload new code the sensor does still fail to initialize, frequently, but it appears you can just ignore this. @PaulZC have you found the same?

Cheers,
Adam

PaulZC, 2021-03-29:
Many thanks Adam,
I haven't tried cycling the power anywhere near as thoroughly as you...
Cheers,
Paul

jerabaul29, 2021-03-29:
@adamgarbo : when doing some testing a couple of months ago I had had similar results that ignoring initialization was helping. However, I was then hitting another problem: I needed to enable BOTH the quaternion and the IMU output, and when some initialization was failing, I was sometimes not able to ask the chip for both of them, and I was getting either the quaternion of the IMU but not both.

Not sure if this works now or not, but that may be a tips for testing: to be really exhaustive, may be a good idea during test to check that you can both:

- start communicating
- set the output to be at least 2 different fields (for example, quaternion and acceleration
- check that the BNO well receives this "instruction" and sends both the quaternion and acceleration data, using for example the ```getReadings()``` function and checking the return code (see for example https://github.com/jerabaul29/Artemis_MbedOS_recipes/blob/e292f421881e37c0444da34585f8d22d57d9ce5a/recipes/recipe_IMU_quaternions/recipe_IMU_quaternions.ino#L81-L134 ).

jerabaul29, 2021-04-09:
@adamgarbo regarding your last test, was it well with the core v1.2.1? Or with the 2.x?

We received a bit of extra funding for building some instruments, but I will have very little time to debug this in the months to come unfortunately. If @PaulZC you have the time to validate how things go with power cycling it would be very helpful.

adamgarbo, 2021-04-09:
Hi @jerabaul29,

I was using Apollo3 Core v1.2.1. The power cycle tests with the MicroMod Artemis Processor and Data Logging Carrier Board were promising but initialization failures and other errors did still regularly occur. However, in most cases, I don't think this prevents the sensor from outputting data.

My main concern is that the sensor has been shown to crash the entire I2C bus and I don't think it's compatible yet with then SAMD-based processors.

jerabaul29, 2021-04-09:
Ok, thanks. I do not think we need to care too much about SAMD processors, we only plan to use Artemis as far as I know.

I know, the I2C crashing is quite annoying. But I think that worst case we will just accept to have some watchog full-reboot.

I think that to survive reboot without issue I will move "important data that should persist and is not used too often" to the flash memory, that will solve all problems.

adamgarbo, 2021-04-09:
I'll continue my testing the BNO080 with SAMD-based processors, as I rely on them for many projects the Artemis is not suitable for (e.g. the Artemis can not make 0-3.3 V analog measurements).

In my opinion, relying on the watchdog timer just to ensure normal operation is a big red flag. The watchdog should really only be used as a last resort to prevent the system from freezing, but of course, it's all a matter of what is acceptable to the end-user.

I will also order an Adafruit BNO085 for testing and continue to explore other IMU options. 

Cheers,
Adam

jerabaul29, 2021-04-09:
Ok, will be very interesting to hear your experience :) .

diplodocuslongus, 2023-10-02:
I get the same behaviour reported in this issue with a BNO086 and a micromod artemis running FW 2.1 in platformio: no IMU detected e.g. after uploading some code or after using the IMU for some (short) time. Before reading this thread I was "restoring"  the BNO086 by power-cycling the micromod ATP board (unplug-replug the USB), but the work by @jerabaul29 and suggestions from @PaulZC point toward alternative options that I may try in the near future.

adamgarbo, 2023-10-02:
@diplodocuslongus I concur with @jerabaul29.

If you wish to continue to use the SparkFun Artemis, save yourself the time and effort and switch to a different IMU . Those are hours, days and weeks of your life you'll never get back! 😉

diplodocuslongus, 2023-10-02:
The previous discussions stopped in 2021 but I see that you guys are still around the corner! Thanks @jerabaul29 for the IMU recommendation, I'll definitely look into it as well as your OpenMetBuoy project (looks very interesting!). Implementing a Kalman Filter on the host MCU is ok. I've tested a few different IMUs over the past 18 months and have been experimenting with the latest BNO086 from CEVA for a month or so, the specs looked impressive. It's still a nice IMU but, well... paired with the artemis results in the same troubles you've experienced.