sparkfun/SparkFun_BNO080_Arduino_Library #74: intermittent, minor issue: non unit quaternions, very high accelerations

jerabaul29, 2021-02-18:
FYI, even grabbing the right kind of packet, I get some intermittent issues with inconsistent quaternions and very high accelerations when polling the sensor at a high frequency. Using my recipe https://github.com/jerabaul29/Artemis_MbedOS_recipes/tree/main/recipes/recipe_IMU_quaternions polling at 200Hz, the output typically looks like what follows. I think these accelerations are not realistic, but I am in a boat with a bit of vibrations - if somebody wants to test on land on a table, that would be great. The quaternions are definitely not consistent - a norm of 2.17 for a rotation quaternion definitely indicates a problem.

This is quite intermittent, remember that I am sampling at 200Hz in this case. So I could just throw away bad values. But still, wanted to report, will be interesting to see if other users see the same, and would be interested to hear if someone has an explanation for it one day :) .

```
************
*HIGH ACCEL*
at ms: 9485, accel vect: vect = [ -126.07, -19.77, -0.62 ]
************
*************
NON UNIT QUAT
at ms: 9676, quat norm: 2.17
*************
************
*HIGH ACCEL*
at ms: 9676, accel vect: vect = [ -3.99, -0.46, 46.64 ]
************
*************
NON UNIT QUAT
at ms: 9680, quat norm: 2.17
*************
************
*HIGH ACCEL*
at ms: 9680, accel vect: vect = [ -3.70, -0.68, 46.29 ]
************
************
*HIGH ACCEL*
at ms: 9731, accel vect: vect = [ -126.09, -19.58, -0.52 ]
************
*************
NON UNIT QUAT
at ms: 9961, quat norm: 0.99
*************
************
*HIGH ACCEL*
at ms: 10035, accel vect: vect = [ -126.02, -19.59, -0.30 ]
************
*************
NON UNIT QUAT
at ms: 10101, quat norm: 1.01
*************
************
*HIGH ACCEL*
at ms: 10231, accel vect: vect = [ -19.46, 126.37, 6.85 ]
************
************
*HIGH ACCEL*
at ms: 10455, accel vect: vect = [ -19.59, 126.30, 6.79 ]
************
*************
NON UNIT QUAT
at ms: 10595, quat norm: 2.17
*************
```

PaulZC, 2021-03-08:
Hi JR (@jerabaul29 ),
I've seen these too. I've captured the I2C traffic at the time of the >|1| event and it looks clean. No bit errors that I can see. Best guess, it is some kind of random event in the Q-Point math? I'll dig a bit deeper on this...
Thanks!
Paul

PaulZC, 2021-03-08:
Yeah. These do look like bit errors in the data. Here's a capture of one event. The I2C logic analyzer says the data was  0x8F 0xFF = -113. Q-Point is 14. -113 * 2^-14 = -0.007. But the code saw 32655 = 0x8F 0x7F (Little Endian) . 32655 * 2^-14 = 1.99. Looks like the MS Bit got corrupted. I'm beginning to suspect the pull-ups?

[image]

PaulZC, 2021-03-08:
I'm now convinced these are bit errors caused by running the bus at 400kHz. I think the same issue causes ```setFeatureCommand``` to fail periodically too. Tweaking the pull-ups _may_ help but I think the only cure is to run at 100kHz...
I don't think there is much more I can do here so I'm going to close this. But please reopen if I've missed something.
Cheers,
Paul

closed 2021-03-08 by PaulZC