sparkfun/SparkFun_BNO080_Arduino_Library #75: output frequency is wrong

jerabaul29, 2021-02-19:
It looks like the output frequency set in code may be wrong.

For example, using the following:

```cpp
#include <Wire.h>

#include "SparkFun_BNO080_Arduino_Library.h" // Click here to get the library: http://librarymanager/All#SparkFun_BNO080
BNO080 myIMU;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("BNO080 Read Example");

  Wire.begin();

  myIMU.begin();

  Wire.setClock(400000); //Increase I2C data rate to 400kHz

  myIMU.enableAccelerometer(50); //Send data update every 50ms

  Serial.println(F("Accelerometer enabled"));
  Serial.println(F("Output in form x, y, z, in m/s^2"));
}

void loop()
{
  //Look for reports from the IMU
  if (myIMU.dataAvailable() == SENSOR_REPORTID_ACCELEROMETER)
  {
    float x = myIMU.getAccelX();
    float y = myIMU.getAccelY();
    float z = myIMU.getAccelZ();

    Serial.print(millis());
    Serial.print(" ");
    Serial.print(x, 2);
    Serial.print(F(","));
    Serial.print(y, 2);
    Serial.print(F(","));
    Serial.print(z, 2);
    Serial.print(F(","));

    Serial.println();
  }
}
```

I get stuff like:

```
3126 1.02,-0.02,9.64,
3157 1.02,0.02,9.61,
3189 1.02,0.02,9.61,
3220 0.83,-0.06,9.61,
3251 1.14,0.02,9.64,
3282 1.14,0.06,9.61,
3313 1.02,0.02,9.61,
3345 1.02,0.02,9.64,
3407 1.02,0.02,9.61,
3439 0.98,0.02,9.64,
3470 1.02,-0.02,9.61,
3501 1.02,0.02,9.64,
3532 1.02,0.02,9.61,
3563 1.06,-0.06,9.61,
3594 0.98,0.06,9.64,
3625 1.02,0.02,9.68,
3657 0.98,-0.02,9.61,
3688 0.98,-0.06,9.57,
3720 1.06,0.10,9.61,
3750 1.02,-0.02,9.64,
3781 1.02,0.02,9.61,
3813 1.02,0.02,9.61,
3844 1.02,-0.02,9.61,
3875 1.02,-0.02,9.61,
3906 1.02,0.02,9.61,
3938 1.02,0.02,9.61,
3969 1.02,-0.02,9.64,
4000 1.02,0.02,9.64,
4031 1.02,-0.52,9.61,
4063 1.02,-0.02,9.64,
4094 0.98,0.02,9.61,
4124 1.02,0.02,9.64,
```

which is not 50ms.

Not sure if this comes from the library, or if it is the chip itself that cannot be trusted, or if I am doing something wrong.

jerabaul29, 2021-02-19:
Honestly, this chip is cursed ^^ .

PaulZC, 2021-03-08:
Hi JR,
The ```setFeatureCommand``` code looks OK.
I wonder if this is quantization caused by the polling of data via the I2C bus?
I wonder if the update interval would be more accurate using SPI and the interrupt pin?
Best wishes,
Paul

PaulZC, 2021-04-06:
Hi JR,
I can't see how this is an issue with the library. I'm going to close this issue as I don't think I can offer any more help - but please reopen if you think I have missed something.
Best wishes,
Paul

closed 2021-04-06 by PaulZC