sparkfun/SparkFun_BNO080_Arduino_Library #76: any experience using UART output?

jerabaul29, 2021-02-22:
Still fighting to the bitter end with my BNO IMUs... ^^ .

Any experience with the UART output? They name 3Mb/s in the datasheet, any idea what baudrate that would correspond to, and if it would be ok to do with the Artemis?

PaulZC, 2021-02-23:
Hi JR,
I haven't tried "Robot Vacuum Cleaner" mode... Your mileage may vary!
The Artemis should be able to cope with 3000000 Baud. But the CH340 Serial-USB interface cannot. You'll need to run the console at a slower rate.
Watch out for serial buffer overruns. The buffer is going to fill up FAST!
The Artemis only has two hardware serial ports. We need one for the console and on AGT the other is dedicated to Iridium. So, I don't know which one you plan to use?
All the best,
Paul

closed 2021-02-23 by PaulZC

jerabaul29, 2021-02-23:
Ok, thanks. Mmmh, I will try first on a Redboard. I have some issues so far, looks like I do not manage to get to 3 000 000 baudrate with the board.