sparkfun/SparkFun_BNO080_Arduino_Library #78: BNO080 not recovering from power cycle (Qwiic Power Switch)

adamgarbo, 2021-02-28:
Hi folks,

Given the issues with the BNO080 and Artemis-based boards, I've been focusing my efforts on using this sensor with the SAMD21 platform (i.e. SparkFun Qwiic Micro).

I've read into the issues with the sleep mode, but my plan was always to use a Qwiic Power Switch (QPS).

The issue I'm encountering is that the BNO080 does not reliably recover from being power cycled with the QPS. I am experiencing the same I2C hang-ups that have been described with Artemis-based boards. When the sensor fails to reinitialize, it causes the entire system to hang (i.e. does not gracefully fail). In addition, hitting the reset button, unplugging the power, or even uploading new code doesn't always seem to fix the issue. 

I've included the barebones example I'm using to test the power cycling using the QPS. It is based on Example17-EulerAngles. Sometimes it works for a cycle or two, but most often it crashes after the first time the QPS disables/enables power.

This makes me very nervous about using this sensor for any real-world projects. I'm keen to get anyone's thoughts on how to improve the code example below. I'm hoping this problem can actually be fixed, though, given all the other issues plaguing the BNO080, I'm worried that it's not looking good.

Cheers,
Adam

```
#include <SparkFun_BNO080_Arduino_Library.h>
#include <SparkFun_Qwiic_Power_Switch_Arduino_Library.h>
#include <Wire.h>

BNO080 myIMU;
QWIIC_POWER mySwitch;

#define Serial SerialUSB

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println("BNO080 Read Example");

  Wire.begin();
  Wire.setClock(400000);

  if (!mySwitch.begin())
  {
    Serial.println(F("Qwiic Power Switch not detected at default I2C address. Please check wiring. Freezing."));
  }

  // Configure IMU
  configureImu();

  // Disable power to IMU
  mySwitch.powerOff();

  // Wait 5 seconds
  delay(5000);
}

void loop()
{
  mySwitch.powerOn();   // Enable power to IMU
  delay(1000);          // Required?
  configureImu();       // Reinitalize IMU
  readImu();            // Read IMU
  mySwitch.powerOff();  // Disable power to IMU
  delay(10000);         // Delay
}

void configureImu()
{
  if (!myIMU.begin())
  {
    Serial.println(F("BNO080 not detected at default I2C address. Freezing..."));
  }

  myIMU.enableRotationVector(50); //Send data update every 50ms
}

void readImu() 
{
  unsigned long loopStartTime = millis();
  while (millis() - loopStartTime < 10000UL)
  {
    // Look for reports from the IMU
    if (myIMU.dataAvailable())
    {
      float roll = (myIMU.getRoll()) * 180.0 / PI; // Convert roll to degrees
      float pitch = (myIMU.getPitch()) * 180.0 / PI; // Convert pitch to degrees
      float yaw = (myIMU.getYaw()) * 180.0 / PI; // Convert yaw / heading to degrees

      Serial.print(roll, 1);
      Serial.print(F(","));
      Serial.print(pitch, 1);
      Serial.print(F(","));
      Serial.print(yaw, 1);

      Serial.println();
    }
  }
}
```

adamgarbo, 2021-03-05:
I believe this issue may be caused by the fact that I'm using a SAMD-based microcontroller.

As I noted on the forums, [here](https://forum.sparkfun.com/viewtopic.php?f=83&t=55233), I've tested several SAMD-based microcontrollers with the BNO080 and experienced the same I2C-related issues with every single one.

1. SparkFun Qwiic Micro (SAMD21)
2. SparkFun MicroMod SAMD51
3. Adafruit Feather M0 (SAMD21)
4. Adafruit Adalogger M0 (SAMD21)
5. Adafruit Feather M0 Express (SAMD21)
6. Moteino M0 (SAMD21)

Even when the QPS is removed from the circuit, the BNO080 does not recover from any form of power cycling. Interestingly, I tested the BNO080 with the nrf52480 MicroMod Processor and was able to power cycle the board several hundred times with absolutely no issues.

PaulZC, 2021-03-08:
Hi Adam,
I _think_ I might have cured this?
Can you please pull the [release_candidate](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/tree/release-candidate) branch and give it a try?
The BNO080 seems to start cleanly on Artemis using I2C whereas it used to fall over regularly. I'd be very interested to see if this works on SAMD?
Thanks!
Paul

adamgarbo, 2021-03-08:
Hi Paul, 

Thanks for looking into this.

I'm afraid no improvements with SAMD-based boards. I can't seem to get either of my BNO080 sensors to initialize even a single time this morning, with or without the QPS.

This really makes me curious as to what microcontrollers *do* work with the BNO080. I saw there's now a Portenta H7 issue created (STM32H747XI).

Cheers,
Adam

adamgarbo, 2021-04-23:
Hi Paul,

I think we can close this issue as outstanding. The BNO080 has a number of major issues but I don't believe the SAMD incompatibility is specific to the SparkFun library, as the same behaviour is experienced with Adafruit's library.

Cheers,
Adam

closed 2021-04-23 by adamgarbo