sparkfun/SparkFun_BNO080_Arduino_Library #80: Problem getting SparkFun VR BNO080 working with the Portenta

cvoalex, 2021-03-07:
I can't for the life of me get this working  I am using the new Portenta H7 and and trying to run the samples for the BNO080.
I am using the Qwiic Cable - Breadboard Jumper (4-pin) to connect to the Portenta.

I get the following:
18:40:29.452 -> BNO080 Read Example
18:40:30.352 -> BNO080 not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...

I have multiple IMUs all with the same issue.  I checked the cables.  Added the delay and flushed the buffer.  I also did the sample I2C scanner and it see the address.  

Any ideas?

PaulZC, 2021-03-07:
Hi @cvoalex ,
Do you have another 3.3V Arduino board you can try - with a different processor?
I'm worried that this might be the same as issue #72 where the BNO080 refuses to play nicely with the Artemis (Apollo3) on I2C.
Try setting the I2C clock frequency to 400kHz - that helps...
Let me know how you get on,
Paul

PaulZC, 2021-03-07:
More info about compatibility problems here:
SAMD21: https://forum.sparkfun.com/viewtopic.php?p=223687#p223687

cvoalex, 2021-03-07:
Hi Paul

I have it working with your Artemis Redboards.

Works fine.  I did the same stuff (delay, flush etc) for the portenta.
Doesnt work.  tried on multiple portentas and multiple VR IMUs.

I even went as far as trying debugging.  It actually sees the address and
receives packets.

I changed the condition to true and it sees all this:

 //Now we wait for response
if (receivePacket() == true)
{
//if (shtpData[0] == SHTP_REPORT_PRODUCT_ID_RESPONSE)
if(1)
{
if (_printDebug == true)
{
_debugPort->print(F("SW Version Major: 0x"));
_debugPort->print(shtpData[2], HEX);
_debugPort->print(F(" SW Version Minor: 0x"));
_debugPort->print(shtpData[3], HEX);
uint32_t SW_Part_Number = ((uint32_t)shtpData[7] << 24) |
((uint32_t)shtpData[6] << 16) | ((uint32_t)shtpData[5] << 8) |
((uint32_t)shtpData[4]);
_debugPort->print(F(" SW Part Number: 0x"));
_debugPort->print(SW_Part_Number, HEX);
uint32_t SW_Build_Number = ((uint32_t)shtpData[11] << 24) |
((uint32_t)shtpData[10] << 16) | ((uint32_t)shtpData[9] << 8) |
((uint32_t)shtpData[8]);
_debugPort->print(F(" SW Build Number: 0x"));
_debugPort->print(SW_Build_Number, HEX);
uint16_t SW_Version_Patch = ((uint16_t)shtpData[13] << 8) |
((uint16_t)shtpData[12]);
_debugPort->print(F(" SW Version Patch: 0x"));
_debugPort->println(SW_Version_Patch, HEX);
}
return (true);
}
}

What doesn't seem to be happening is anythimg writing to the header
shtpData.

I appreciate any help you can give
Tx

cvoalex, 2021-03-07:
01:27:22.743 -> Request ProductID
01:27:22.743 -> TRUE  Received packet
01:27:22.743 -> SW Version Major: 0x3 SW Version Minor: 0x2 SW Part Number:
0x98A498 SW Build Number: 0x172 SW Version Patch: 0x7
01:27:22.743 -> Received Packet1
01:27:22.743 -> 0000Packet 0000
01:27:22.743 -> Readings1

PaulZC, 2021-03-07:
Hi @cvoalex ,
I suspect you could end up going down a rabbit hole here...
If you have an (inexpensive) [logic analyzer](https://www.sparkfun.com/products/15033) you can look at the I2C traffic and try and work out what is going on. But brace yourself for weirdness. I spent _hours_ looking at the Artemis (Apollo3) issue and couldn't solve it. It crashes the Apollo3 I2C hardware at a very low level causing it to timeout. You see normal bus traffic and then it just stops dead with no clues as to why...
I'll leave this issue open for a while in case anyone else can help.
All the best,
Paul

cvoalex, 2021-03-07:
Paul

Seems these guys got it working.  Dont know how

https://github.com/adafruit/Adafruit_BNO08x

Maybe you guys can do the same.
Tx
Alex

PaulZC, 2021-03-07:
Hi Alex,
So have you tried Adafruit's library? Does it work on your Portenta?
Thanks,
Paul

cvoalex, 2021-03-07:
Hi Paul 
Yes.  Tried it and it works

Tx

PaulZC, 2021-03-07:
Hi Alex,
Perfect- thanks! I might stand a chance of debugging it now - if the same hardware works with the Adafruit library but doesn’t work with the SF version.
All the best,
Paul

cvoalex, 2021-03-07:
Great. Good luck

See their issue thread. Someone about some buffer >32

cvoalex, 2021-03-08:
I commented out this line

    //while (1) {     delay(10);   }

So there is clearly still some problem with the begin, but otherwise works

PaulZC, 2021-03-08:
Hi Alex,
If you're familiar with GitHub (Git) branches, can you please pull or download the [release_candidate branch](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/tree/release-candidate) and give it a try?
I've made some changes there and it now seems to start reliably on Artemis using I2C. I'd be very interested to see if it works on Portenta too.
Many thanks!
Paul

cvoalex, 2021-03-09:
Hi Paul

Unfortunately no luck.

It throws the Portenta into panic mode.
Tx
Alex

cvoalex, 2021-03-26:
Hi Paul

Still trying to get the VR BNO going.

I connected it to the Redboard and got it working fine with I2C.

But I cant get it going with the SPI.  I connected the pins per the SPI
example to the appropriate pins on the breakout board.

Any ideas?

PaulZC, 2021-03-26:
Hi Alex,
Are there any clues about where the Portenta goes into panic? Can you trace it to a line of code? (I'm wondering if I have done something silly like doing a debug print when the debug serial port has not been specified.)
The BNO does work fine on SPI on a RedBoard, but you do need to follow the instructions carefully. Please check you have closed the PS1 jumper and have opened both sides of the the I2C jumper.
https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/examples/SPI/Example1-RotationVector/Example1-RotationVector.ino#L20-L36
Cheers,
Paul

cvoalex, 2021-03-26:
Paul

Thanks for getting back.  I did exactly as asked.  No luck.

SPI Wait
SPI Wait
SPI Wait
SPI Wait
SPI Wait
SPI Wait
SPI Wait
SPI Wait
SPI Wait
SPI INT timeout
BNO080 over SPI not detected. Are you sure you have all 6 connections?
Freezing...

This is the code:
/*
  Using the BNO080 IMU
  By: Nathan Seidle
  SparkFun Electronics
  Date: July 27th, 2018
  SparkFun code, firmware, and software is released under the MIT License.
  Please see LICENSE.md for further details.
  Feel like supporting our work? Buy a board from SparkFun!
  https://www.sparkfun.com/products/14686
  This example shows how to use the SPI interface on the BNO080. It's
fairly involved
  and requires 7 comm wires (plus 2 power), soldering the PS1 jumper, and
clearing
  the I2C jumper. We recommend using the Qwiic I2C interface, but if you
need speed
  SPI is the way to go.
  This example shows how to output the i/j/k/real parts of the rotation
vector.
  https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation
  Hardware modifications:
  The PS1 jumper must be closed
  The PS0 jumper must be open. PS0/WAKE is connected and the WAK pin is
used to bring the IC out of sleep.
  The I2C pull up jumper must be cleared/open
  Hardware Connections:
  Don't hook the BNO080 to a normal 5V Uno! Either use the Qwiic system or
use a
  microcontroller that runs at 3.3V.
  Arduino 13 = BNO080 SCK
  12 = SO
  11 = SI
  10 = !CS
  9 = WAK
  8 = !INT
  7 = !RST
  3.3V = 3V3
  GND = GND

  black = GND
  grey = SCK
  purple = MISO
  blue = MOSI
  green = 10 = CS
  yellow = 9 = wake
  orange = 8 = int
  red = 7 = reset

*/

#include <SPI.h>

#include "SparkFun_BNO080_Arduino_Library.h" // Click here to get the
library: http://librarymanager/All#SparkFun_BNO080
BNO080 myIMU;

//These pins can be any GPIO
byte imuCSPin = 10;
byte imuWAKPin = 9;
byte imuINTPin = 8;
byte imuRSTPin = 7;

unsigned long startTime; //Used for calc'ing Hz
long measurements = 0; //Used for calc'ing Hz

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("BNO080 SPI Read Example");

  myIMU.enableDebugging(Serial); //Pipe debug messages to Serial port

  if(myIMU.beginSPI(imuCSPin, imuWAKPin, imuINTPin, imuRSTPin) == false)
  {
    Serial.println("BNO080 over SPI not detected. Are you sure you have all
6 connections? Freezing...");
    while(1);
  }

  //You can also call begin with SPI clock speed and SPI port hardware
  //myIMU.beginSPI(imuCSPin, imuWAKPin, imuINTPin, imuRSTPin, 1000000);
  //myIMU.beginSPI(imuCSPin, imuWAKPin, imuINTPin, imuRSTPin, 1000000,
SPI1);

  //The IMU is now connected over SPI
  //Please see the other examples for library functions that you can call

  myIMU.enableRotationVector(50); //Send data update every 50ms

  Serial.println(F("Rotation vector enabled"));
  Serial.println(F("Output in form i, j, k, real, accuracy"));

  startTime = millis();
}

void loop()
{
  Serial.println("Doing other things");
  delay(10); //You can do many other things. We spend most of our time
printing and delaying.

  //Look for reports from the IMU
  if (myIMU.dataAvailable() == true)
  {
    float quatI = myIMU.getQuatI();
    float quatJ = myIMU.getQuatJ();
    float quatK = myIMU.getQuatK();
    float quatReal = myIMU.getQuatReal();
    float quatRadianAccuracy = myIMU.getQuatRadianAccuracy();
    measurements++;

    Serial.print(quatI, 2);
    Serial.print(F(","));
    Serial.print(quatJ, 2);
    Serial.print(F(","));
    Serial.print(quatK, 2);
    Serial.print(F(","));
    Serial.print(quatReal, 2);
    Serial.print(F(","));
    Serial.print(quatRadianAccuracy, 2);
    Serial.print(F(","));
    Serial.print((float)measurements / ((millis() - startTime) / 1000.0),
2);
    Serial.print(F("Hz"));

    Serial.println();
  }

}

[image: IMG_6776.jpg][image: IMG_6771.jpg][image: IMG_6772.jpg][image:
IMG_6773.jpg][image: IMG_6775.jpg]

cvoalex, 2021-03-27:
Hi Paul

Online on your site the SPI jumpers are different.

[image: image.png]

PaulZC, 2021-03-27:
Hi Alex,
Your image didn't make it through. Which SPI jumpers?
Thanks,
Paul

adamgarbo, 2021-03-27:
Interesting that the STM32H747 experiences the same I2C issues with the BNO080. I had thought it supported clock-stretching out of the box. So far, the only processor I've had complete success using with the BNO080 is Nordic's nRF52480.

If I recall correctly, wasn't the reason for the BNO085 update specifically because of SPI-related issues?

I'm tempted to try to use the BNO080 over serial, but I worry it's just another rabbit-hole.

cvoalex, 2021-03-27:
IMG_6771.jpg
<https://drive.google.com/file/d/1EAVjy7z-hoNEWQPY7U09pC_hzdRTkh_f/view?usp=drive_web>

 IMG_6772.jpg
<https://drive.google.com/file/d/1O5qlZPR_Cw8978Qa1zO9yzY0y9jK-UFr/view?usp=drive_web>

 IMG_6773.jpg
<https://drive.google.com/file/d/1USsC3BF7J8k3N2c_ZU1di8HQeyspFd8U/view?usp=drive_web>

 IMG_6775.jpg
<https://drive.google.com/file/d/154ly0r0i0w6y_wV5xBUuZ78Hh4OZv8JF/view?usp=drive_web>

 IMG_6776.jpg
<https://drive.google.com/file/d/1-s0nN_5rMZTauWLeFxW19RsADNnzrkcZ/view?usp=drive_web>

PaulZC, 2021-03-28:
Thanks Alex,
Please try clearing the solder from the ADR split pad. ADR is connected to the SI (SPI In) pin. The solder will be holding the line low.
If this works - and I'm sure it will - I'll add a note about it in the SPI examples.
Best wishes,
Paul

cvoalex, 2021-03-28:
Paul

Thank you. I wont be able to try until Wednesday as I have gone out of town. Will let you know how it goes

Take care
Alex

PaulZC, 2021-03-28:
No problem Alex,
When you get back, please try the release_candidate branch again on Portenta. I had made a silly mistake with the debug prints...
https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/commit/9e983ded24001e475d0cccbf43e8bffecc8a41d9
Fixed now.
Many thanks for your help and patience,
Paul

cvoalex, 2021-04-01:
> 9e983de

Hi Paul
I have it working well on the RedBoard.

Unfortunately, doesnt work on the Portenta at all.  I am assuming it has
something to do with Portenta.

Do you guys have a portenta there to try it against?
Tx

PaulZC, 2021-04-01:
Thanks Alex,
Unfortunately I don't have a Portenta...
Has the Portenta "panic" gone away? What error do you see now?
Thanks,
Paul

cvoalex, 2021-04-01:
No Portenta still has issues  Seems issues with generic SPI.

But works fine on Redboards.

OK Will keep looking.

BTW?  Any way I can help pay you guys to help figure this out?  Does
Sparkfun offer such a service?
Tx