sparkfun/SparkFun_BNO080_Arduino_Library #81: Euler, Quats and refresh rate questions

ndanilo8, 2021-03-26:
### Subject of the issue

.getRoll() .getpitch() .getyaw() will return the respective euler angles 
So I've been questioning myself about this functions. I've checked the .cpp and those are calculated from the quaternions pulled from the rotation vector mode, correct?

I wonder if they are gimbal lock compensated? As for the project I'm working I need to know the relative/absolute orientation and then push that data(angles) to some servos for control. So can I use this data straight away? Or is it better to convert the quaternions to Euler angles myself?

Also there is this  functions called in setup() eg: `myIMU.enableRotationVector(50); //Send data update every 50ms` can this be considered as the refresh rate of the sensor? and if so what is the fastest refresh data I can pull from the sensor? 10ms? 

Thank you in advance 

### Your workbench
* Teensy 4.1
* Adafruit BNO085 breakout board
* via I2C
* LiPo Battery

PaulZC, 2021-03-26:
Hi Danilo (@ndanilo8 ),
Correct - roll, pitch and yaw are calculated from the quats:
https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/src/SparkFun_BNO080_Arduino_Library.cpp#L415-L489
They are _not_ gimbal lock compensated. I would recommend working directly with the quats if you can.
Please check the BNO080 datasheet for the fastest rate. It may depend on which interface you are usuing. SPI is way faster than I2C.
Best wishes,
Paul

closed 2021-03-26 by PaulZC

ndanilo8, 2021-03-28:
Hey @PaulZC ,
Thanks for your fast input, I appreciate it.
I have another question:
Should I get the accel, gyro, mag directly and use a AHRS filter like Madgwick/mahony to get to the euler angles ? But I wonder if I'm loosing the "magic"/(Hilcrest fusion algorithm) away from BNO as it already gives me the proceed quats and perhaps is it best for me to use that? Or the first option with the Madgwick/mahony filter?

I also changed the circuitry to work with SPI protocol instead of I2C for higher refresh rates. 
[image]
I still have a question regarding the `myIMU.enableRotationVector(50);` as its written on the datasheet the max for rotation vector is 400 Hz so should I set this like so? : `myIMU.enableRotationVector(400);` Or will this in fact be slower than (50)? Kinda confused...

PaulZC, 2021-03-28:
Hi @ndanilo8 ,

You are welcome to try adding your own filters but I suspect you will simply be replicating what the BNO does internally.

```enableRotationVector``` calls ```setFeatureCommand``` which takes the time between reports / report interval in milliseconds as a parameter. The milliseconds get converted to microseconds inside ```setFeatureCommand```.
So ```myIMU.enableRotationVector(50);``` sets the rate to 20Hz (50ms report interval).

Cheers,
Paul