sparkfun/SparkFun_BNO080_Arduino_Library #82: BNO080 not working on shared SPI bus (multiple SPI slaves)

SolidGeek, 2021-04-06:
### Subject of the issue
I got the BNO080 and a PMW3901 sensor that I would like to communicate on the same SPI bus. Both devices have their own chip-select pin CS1 and CS2, and works just fine independently. However, when I connect the PMW3901's MISO line to the SPI port on which the BNO080 is connected, the BNO080 stops working. I have enabled debugging, and it seems that my MCU receives an empty header after the first H_INT is received (at setup). However, after that, no more data are received at the BNO080 stops talking. I have tried analysing the communication with my digital oscilloscope, but I can't seem to figure out the problem. Are the BNO080 not capable of sharing an SPI port?

### Your workbench
I am using a Teensy 4.0 as the MCU, and the PIM453 breakout board for the PMW3901 sensor. Everything is powered by a 3.3V external LDO. The BNO080 board I am using is the FSM-300 board, wired as follows:
[image]

### Steps to reproduce
The sensors and Teensy are connected as follows: 
[image]

This is the code I used:
```c++
#include "src/BNO080.h" 
#include "src/PMW3901.h"

#define FLOW_CS_PIN 6
#define IMU_CS_PIN 10
#define IMU_WAK_PIN 22
#define IMU_INT_PIN 21
#define IMU_RST_PIN 20

BNO080 IMU;
PMW3901 flow(FLOW_CS_PIN);

void setup() {
  pinMode( FLOW_CS_PIN, OUTPUT );
  digitalWrite(FLOW_CS_PIN, HIGH);
  
  Serial.begin(115200);
  
  IMU.enableDebugging(Serial);
  if(IMU.beginSPI(IMU_CS_PIN, IMU_WAK_PIN, IMU_INT_PIN, IMU_RST_PIN) == false){
    Serial.println(F("BNO080 over SPI not detected."));
    while(1);
  } 

  delay(1000);

  if (!flow.begin()) {
    Serial.println("Initialization of the flow sensor failed");
    while(1);
  }
}

void loop() {
  // Nothing
}
```

### Expected behavior
I would expect that both sensors can communicate on the same SPI bus, when being selected using their respective CS pin. 

### Actual behavior
Only the PMW3901 can communicate when both sensors are connected on same SPI bus. Maybe the setup of the BNO080 fails when another SPI device is present? 

My project is short on GPIO pins, thus I cannot use a separate SPI port for the PMW3901. Can you help me figure out why both devices can't communicate on the same SPI port?

PaulZC, 2021-04-06:
Hi Emil (@SolidGeek ),
GitHub repo issues are not the best place to ask for technical help. You have a hardware issue - not an issue with the library. You can try asking for help on the [SparkFun forums](https://forum.sparkfun.com/viewforum.php?f=83) if you wish.
Best wishes,
Paul

closed 2021-04-06 by PaulZC

SolidGeek, 2021-04-06:
Hi Paul, thanks. You are right, however I am still interested in knowing, if you have tested the BNO080 SPI communication, with another SPI slave present on the bus? This would help me alot.

PaulZC, 2021-04-06:
I would recommend buying a [cheap logic analyzer](https://www.sparkfun.com/products/15033) and use it to monitor all of your bus pins. You need to make sure that you do not have "bus contention" when two devices are trying to communicate simultaneously. Especially, check that there is no overlap on your CS signals.
Maybe you can configure your oscilloscope to trigger if both CS signals are low?
Good luck with your project,
Paul

75snow75, 2025-01-08:
Hi

I faced same problem with you...
Did you solve it?