sparkfun/SparkFun_BNO080_Arduino_Library #83: Please could you implement the TARE function? Thank you!

michroz2, 2021-04-08:
### Please could you implement the TARE function? Thank you!
The BNO08X have Tare function described in the "SH-2 Reference Manual Document Number: 1000-3625" and
"1000-4045 - App Note BNO080/BNO085 Tare Function Usage Guide"
I find this feature useful for the cases of BNO08X  installations independent of the ground level.
I actually implemented this function for me, but for my specific case of Game Vector and XYZ axes only, while for the public library it probably should feature all possible bells and whistles.

### Your workbench
*irrelevant

### Steps to reproduce
irrelevant

### Expected behavior
The BNO08X should "Tare", i.e. re-orient their vectors according to the current actual position of the local G and M vectors. 

### Actual behavior
No such function in the library

###Below is my humble implementation of the Tare (based on version 1.1.9)
.h file:
	void sendCommand(uint8_t command);
	void sendTareGameXYZCommand();	//MR> I added this command to this library for my project - by example.
	void sendCalibrateCommand(uint8_t thingToCalibrate);
.cpp file:
//MR> This tells the BNO080 to Tare
//MR> See page 45 of reference manual
void BNO080::sendTareGameXYZCommand()
{
	/*Setting the shtpData[]
	byte Name Description
	3	 P0	  0x00 – Subcommand: Perform Tare now
	4	 P1	  Bitmap of axes to tare: Bit 0= X, Bit 1= Y, Bit 2= Z		MR> 7 = X,Y,Z
	5	 P2	  Rotation Vector to use as basis for tare.
				0: Rotation Vector
				1: Gaming Rotation Vector								MR> This!
				2: Geomagnetic Rotation Vector
				3: Gyro-Integrated Rotation Vector
				4: ARVR-Stabilized Rotation Vector
				5: ARVR-Stabilized Game Rotation Vector
	6	 P3	 Reserved
	7	 P4	 Reserved
	8	 P5	 Reserved
	9	 P6	 Reserved
	10	 P7	 Reserved
	11	 P8	 Reserved
	*/

	for (uint8_t x = 3; x < 12; x++) //Clear this section of the shtpData array
		shtpData[x] = 0;

	shtpData[4] = 7; //all axes
	shtpData[5] = 1; //game rotation vector

	//Using this shtpData packet, send a command 0x03
	sendCommand(COMMAND_TARE);
}

//This tells the BNO080 to begin calibrating
//See page 50 of reference manual and the 1000-4044 calibration doc
void BNO080::sendCalibrateCommand(uint8_t thingToCalibrate)
{
...

PaulZC, 2021-04-08:
Hi Michael (@michroz2 ),
This is a reasonable request, but we will not be able to implement it any time soon.
If you want to send us a Pull Request, that would be great.
Best wishes,
Paul

michroz2, 2021-04-09:
Thanks Paul!
Maybe I will.
I am just a hobbyist programmer though.
Best to you,
Mich

sofian, 2022-03-01:
I have implemented it and would be happy to provide a pull request. What would be the preferred API?
Tare has a number of options, namely:
* "tare now" with options for "z axis only" or "all axis"
* "tare persist" which makes the tare persistent

How about something like this:
```c++
void tare(bool zAxis=false); // <-- tare now
```
Or:
```c++
void tareNow(bool zAxis=false); // <-- tare now
```

Or:
```c++
void tareZ(); // <-- tare now according to z-axis
void tareAll(); // <-- tare now according to all axis
```

For the "persist" I would suggest:
```c++
void saveTare(); // <-- tare persist
```

michroz2, 2022-03-02:
My vote is for the second option. Thank you!

michroz2, 2022-03-03:
> Just did it, let me know what you think.

I will, but I'm only able at the next update of my project, happening,
hopefully this year.
Did you use this function in your project or what was your incentive for
doing this update?
Best and Thank You!

sofian, 2022-03-19:
I needed it for one of my own projects. No problem about the update, just happy to contribute.

PaulZC, 2023-01-23:
Hi Michael (@michroz2 ),

Apologies for the very slow reply. I have merged @sofian 's code. I will up-issue the library once I have tested it.

Closed by #99 .

Best wishes,
Paul

closed 2023-01-23 by PaulZC

michroz2, 2023-01-23:
Thank you Paul!