sparkfun/SparkFun_BNO080_Arduino_Library #85: disable previously enabled functions, e.g. stepCounter

marcel-bluestone, 2021-05-19:
Hello,
is it possible to disable specific features such as the stepCounter if it is no longer needed, but other features such as GameRotationVector should be activated instead?

If not, should I do a softReset and create a new I2C connection instead? I am curious what you would recommend?

PaulZC, 2021-05-19:
Hi @marcel-bluestone ,
What happens if you call ```enableStepCounter(0);``` (i.e. set the time between reports to zero)?
Best wishes,
Paul

marcel-bluestone, 2021-05-19:
Hi @PaulZC 

I will try that at home tonight. I was thinking about this, but was afraid this might end up in an eternal loop.
I will report if this was successful. 
Thanks for the hint.

Marcel

marcel-bluestone, 2021-05-20:
this worked, thank you @PaulZC !

closed 2021-05-20 by marcel-bluestone