sparkfun/SparkFun_BNO080_Arduino_Library #87: setFrs() function for setting specific config parameters

marcel-bluestone, 2021-05-24:
Hi, I want to use the shakeDetector functionality of the BNO080 with custom parameters.
The sh2 reference manual gives instructions to set a 5 WORD array custom configuration under (4.3.7) using the setFrs function. Is this somehow possible to set with the current sparkfun lib? I found that adafruit had implemented this in their lib, which unfortunately is able to support only one device at a time.
I do need to use two BNO080 devices in parallel via I2C so I'd rather stick with this lib.

PaulZC, 2021-05-24:
Hi @marcel-bluestone ,
I will try and find time to look at this, but I have many other things to do first. If you need a solution soon, it might be best to continue with the Adafruit library.
Best wishes,
Paul

badVibes--, 2023-03-19:
You can check out my branch. I implemented writing to FRS and made a method to configure the giro integrated rotation vector. Also did some other improvements. Specifically I fixed SPI initialisation which would cause the hang when the chip would reset. 

My branch is behind since this was several years back, otherwise I would submit a pull request.