sparkfun/SparkFun_BNO080_Arduino_Library #89: SoftWire isn't compatible

EmDash00, 2021-06-16:
In the examples, it mentioned we could possibly use SoftWire; however, it seems that SoftWire doesn't inherit from TwoWire, which makes the typing unhappy. You might want to change the example to mention this.

PaulZC, 2021-06-17:
Hi Drason (@EmDash00 ),
Which platform / board are you using?
Best wishes,
Paul

PaulZC, 2021-06-17:
Never mind... Fixed in [this commit](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/commit/f318297613a7dbde485c1b4fb20fe7d39cc22a71).
We could add support for SoftWire but it would mean adding overloaded of #ifdef begin functions etc.. And I'm not convinced that SoftWire would work well on the BNO080 anyway. The BNO can be tricky on I2C...
Closing...
Paul

closed 2021-06-17 by PaulZC