sparkfun/SparkFun_BNO080_Arduino_Library #93: Full rotation is 2 degrees short of 360

nathanWagenbach, 2021-08-19:
When I rotate my BNO080 exactly 360 degrees it is always about 2 degrees short each rotation. For example, if I rotate it 360 degrees it will say the yaw is 358 degrees. If I rotate 720 the yaw would be 716. Also, If i rotate to 360 and back to 0 it will have the correct yaw of 0. Is this expected  behavior for the BNO080?  I have tried lots of different examples from the sparkfun library and the results are the same. This also occurs in the rotation vector, game rotation vector, and gyro integrate rotation vector.

Development board: Teensy 3.5
Breakout Board: Sparkfun VR IMU breakout
Connection: I2C

PaulZC, 2021-08-19:
Hi @nathanWagenbach ,

The yaw angle is calculated from the Quaternion values here:

https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/f318297613a7dbde485c1b4fb20fe7d39cc22a71/src/SparkFun_BNO080_Arduino_Library.cpp#L467-L489

and is then converted from Radians to Degrees in the example:

https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/f318297613a7dbde485c1b4fb20fe7d39cc22a71/examples/Example17-EulerAngles/Example17-EulerAngles.ino#L75-L77

Yaw is calculated using ```atan2``` so, yes, I would expect this behavior. The yaw will get _close_ to 2.Pi radians (360 degrees) and then wrap round to zero.

Euler angles calculated like this will always be an approximation - and will suffer from gimbal lock. I would recommend working directly with the Quaternions if you can. 

I'm closing this issue as I believe the library and example are working as expected. But please re-open if I have missed something.

Best wishes,
Paul

closed 2021-08-19 by PaulZC

nathanWagenbach, 2021-08-19:
Thanks for the quick response @PaulZC

Unfortunately I get the same issue when using Quaternion output directly.
For example, when I first power up my board with the game rotation vector enable the output quaternion is:
[-0.01, -0.00, 0.01, 1.00, 0.25]. The format is [quatI, quatJ, quatK, quatReal, quatAccuracy]. 
When I rotate the board 360 degrees ccw the values are:
[-0.01, 0.00, -0.02, 1.00, 0.25]
If I rotate another 360 degreees ccw the values are:
[0.01, -0.01, -0.04, 1.00, 0.25]
and as I continue rotating ccw:
[-0.01 ,0.01, -0.06, 1.00, 0.25]
[-0.01, 0.01, -0.08, 1.00, 0.25]
[-0.01, 0.00, -0.10, 1.00 ,0.25]
When I rotated it all the way back to its starting position it was almost exactly correct at:
[-0.00,-0.00,0.00,1.00,0.25]

As you can see the K quaternion decreases by about 0.02 each rotation. However, it does not drift when static. I cannot seem to find the source of these issues. I could see a magnetometer throwing them off but as far as I know the game rotation vector does not use the magnetometer.

Thanks,
Nathan

PaulZC, 2021-08-19:
Hi @nathanWagenbach ,

Please have a read through issue #49. I think you might be seeing the same issue?

Best wishes,
Paul

PaulZC, 2021-08-20:
After thinking about this overnight, I'm pretty sure this is a gyro issue. If the gyro calibration is slightly off, you would - I think - see this effect.

Please make sure you are giving the BNO080 time to internally calibrate itself before you try your test. You need to rotate it around all three axes, and hold it stationary in all six orientations for a few seconds. That lets the internal processor calculate the gain and bias/offset of all the sensors.

If you are still having problems, you could ask your question in the [IMU Forum](https://forum.sparkfun.com/viewforum.php?f=83). Some of the contributors there know way more about IMUs than I do.

Best wishes,
Paul