sparkfun/SparkFun_BNO080_Arduino_Library #100: SPI Communication not working outside of Arduino IDE

adamoskaranikas, 2022-03-11:
### Subject of the issue
Describe your issue here.

### Your workbench
* What development board or microcontroller are you using?
ESP32 Thing Plus v2.0
* What version of hardware or breakout board are you using? 
Sparkfun BNO080 breakout board.
SW Version Major: 3 SW Version Minor: 2 SW Part Number: 0x98A498 SW Build Number: 172 SW Version Patch: 7 

* How is the breakout board to your microcontroller?
SPI
* How is everything being powered?
USB to the ESP, 3v3 from the esp to the imu

* Are there any additional details that may help us help you?
I have been using the BNO with ESP-IDF's Arduino-ESP32Core with this driver, but connected in I2C, and managed to get very good performance out of the board (~75% of each sensor's max reported rate, aside from the rotation vectors which I got a max of around 350Hz only instead of close to 1kHz). I decided to switch to SPI communication in order to test out if I2C was the limiting factor in speed. I started with using the arduino IDE and one of the SPI example scripts, and it worked, so I moved my code in ESP-IDF from I2C to SPI, and now I only receive the first packet after reset: 
` SW Version Major: 3 SW Version Minor: 2 SW Part Number: 0x98A498 SW Build Number: 172 SW Version Patch: 7 `
and no sensor data is coming through.
### Steps to reproduce
This is a stripped down version of the code i've been running.
```

#include "Wire.h"
#include "SparkFun_BNO080_Arduino_Library.h"
#include <string.h>
#include <stdlib.h>
#include <stdio.h>
#include <unistd.h>
#include <SPI.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "nvs.h"
#include "nvs_flash.h"

#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"

#define INT1 (32)

// defining SPI Pins
#define SCK 18
#define MISO 19
#define MOSI 23
#define CS 5
#define WAK 22
#define RST 21

SPIClass SPI2IMU;
BNO080 moto_imu;

typedef struct
{
    float x;
    float y;
    float z;
    float w;
    float radAcc;
    uint8_t accuracy;
} quat;

typedef struct
{
    float x;
    float y;
    float z;
    uint8_t accuracy;
} xyz;

extern "C" void imu_task(void *ignore) {
	
    
    moto_imu.enableDebugging();

    ESP_LOGW(TAG, "Trying beginSPI: %d", moto_imu.beginSPI(CS, WAK, INT1, RST, 1000000, SPI2IMU, SCK, MISO, MOSI));

    ESP_LOGI(TAG, "Enabling Sensors..");

    moto_imu.enableLinearAccelerometer(100);
    // moto_imu.enableRawGyro(50);
    // moto_imu.enableGyroIntegratedRotationVector(100);
    // moto_imu.enableMagnetometer(100);
// i've commented the other sensors out but normally i'd be enabling at least 3 of these

    ESP_LOGW(TAG, "LinAccel on");
    xyz gyro, accel, mag; 
    quat q;
	for(;;)
            if (moto_imu.dataAvailable()){
                // ESP_LOGI(TAG, "Exited indef. block, rc: %d", gpio);
                uint16_t sensor = moto_imu.getReadings();
                
                // ESP_LOGW(TAG, "RECV Sensor: %u ", sensor);
                switch (sensor)
                {
                case SENSOR_REPORTID_LINEAR_ACCELERATION:
                    printf("a");
                    moto_imu.getLinAccel(accel.x, accel.y, accel.z, accel.accuracy);
                    break;
                case SENSOR_REPORTID_GYRO_INTEGRATED_ROTATION_VECTOR:
                    printf("q");
                    moto_imu.getQuat(q.x, q.y, q.z, q.w, q.radAcc, q.accuracy);
                    break;
                case SENSOR_REPORTID_RAW_GYROSCOPE:
                    printf("g");
                    moto_imu.getFastGyro(gyro.x, gyro.y, gyro.z);
                    gyro.accuracy = moto_imu.getGyroAccuracy();
                    break;
                case SENSOR_REPORTID_MAGNETIC_FIELD:
                    printf("m");
                    moto_imu.getMag(mag.x, mag.y, mag.z, mag.accuracy);
                    break;

                case 0:
                    printf("getReadings: 0 \n");
                    vTaskDelay(100);
                    break;
                default:
                    printf("smth else: %u -----------", sensor);
                    break;
                }

                
            }
            else{
                ESP_LOGW(TAG, "Waiting for data.. ");
                         vTaskDelay(80);
                
            }

	}
	vTaskDelete(NULL);
}

extern "C" void app_main(void)
{
   gpio_reset_pin(LED);
   gpio_set_direction(LED, GPIO_MODE_OUTPUT);
   gpio_set_level(LED, 1);

    // Initialize NVS.
    esp_err_t err = nvs_flash_init();

    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // 1.OTA app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // 2.NVS partition contains data in new format and cannot be recognized by this version of code.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    xTaskCreate(&imu_task, "imu_task", 1024*8, NULL, 10, &imuTask);  

    
}

```

I've also edited the cpp file in order for enable debugging to print complying to esp-idf, and  `beginSPI` function to take in the MISO, MOSI, SCK signals in order to begin the `_spiPort` with the right pins. The esp32 is quite specific with which pins you can use wrt the GPIO MUX and getting the best performance out.

### Expected behavior

getReadings() should give back a sensor ID and then switch case fills up my structs with the data every loop.  (as a side note, getReadings() works fine when the sensor is configured in i2c.)

### Actual behavior

beginSPI returns true, the first advertisement packet is received and printed, and then on each instance of the intr line going low, and dataAvailable() returning true, getReadings() returns 0.

----
I've tried getting this to work for about a week now and am not making any progress. I'm also writing my own driver for the BN0 using the ESP-IDF spi drivers but that is also a huge mess, which will take a long time,  so I'm hoping I can resolve this in the meantime in order to test performance. 

I'm not sure what it means when getReadings() returns 0, I'm guessing that something goes wrong in the spi transfer, but would love to get some info from you guys about it.

PaulZC, 2022-03-11:
Hi @adamoskaranikas ,

You're unlikely to get any help with this issue here... You appear to be having an issue with your ESP-IDF code, not this library. But I'm happy to leave this issue open for a while, just in case anyone can help.

You may have more luck asking your question in the SparkFun IMU Forum:
https://forum.sparkfun.com/viewforum.php?f=83

Best wishes,
Paul

adamoskaranikas, 2022-03-17:
Hi @PaulZC , thanks for the pointer, I've managed to get the sparkfun lib to work for SPI within the IDF, it ended up needing just a slightly different initialisation. I'll close the issue. I still foresee some problems coming up however haha but we'll see. 

Thanks!

closed 2022-03-17 by adamoskaranikas

VICLER, 2024-08-23:
> Hi @PaulZC , thanks for the pointer, I've managed to get the sparkfun lib to work for SPI within the IDF, it ended up needing just a slightly different initialisation. I'll close the issue. I still foresee some problems coming up however haha but we'll see.
> 
> Thanks!

Hi @adamoskaranikas !
Do you remember how you changed the SPI initialization? I am working with this library and bno085 sensor and I get good results using i2c, but I can not get it working with SPI.